schreibfaul1/ESP32-audioI2S #476: How to use switch internet radio to bluetooth

bhpjean, 2023-02-03:
Is there any chance using this software to be able to switch from wifi to bluetootch and send sound from the phone? Or how to make it possible to switch internet radio / bluetootch receiver

biologist79, 2023-02-03:
Sounds like you want to have a look at [ESPuino](https://github.com/biologist79/ESPuino) 😃

bhpjean, 2023-02-04:
> Sounds like you want to have a look at [ESPuino](https://github.com/biologist79/ESPuino) smiley

Oops This is too complicated for my knowledge :-)
I already have a project for my internet radio and I would like to develop it. The project is based on the software available here and my joyful creation :-)
In this equipment I wanted to run BT adding only a few lines of code if possible and preferably in the Arduino IDE
https://www.youtube.com/watch?v=LGwflOgD88s

schreibfaul1, 2023-02-04:
It's a nice project, I like it. The developers of the ESPuino project know very well about BT. You can ask questions in the forum and you will surely get a competent answer. There is an interface for extracting the audio data from the library. Unfortunately, it takes more than just a few lines of code.
As a replacement, you can use a BT transmitter/receiver without any additional code, e.g. this one:
[image]
With the model, the housing can be opened easily, it is only clipped. Four wires to the DAC and BT is ready.

renatoianhez, 2023-02-05:
Hi! On the [radio](https://github.com/renatoianhez/Lolin32-radio) I built, I put an SPDT switch that switches between bluetooth and radio modes. It worked great!

rv9c, 2023-02-08:
Hi all. BT audio can be implemented without third-party libraries like this, everything is very simple.

`// cricial audio bits taken from https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/main.c

// bluetooth, config, discover and audio
#include "esp_bt_main.h"
#include "esp_bt_device.h"
#include "esp_gap_bt_api.h"
#include "esp_a2dp_api.h"

// the audio DAC and amp configuration. 
#include "driver/i2s.h"

//globals

// the callback(processes bluetooth data).
// this is the most important function. 
void bt_data_cb(const uint8_t *data, uint32_t len){
   // number of 16 bit samples
   int n = len/2;
   
   // point to a 16bit sample 
   int16_t* data16=(int16_t*)data;
   
   // create a variable (potentially processed) that we'll pass via I2S 
   int16_t fy; 
   
   // Records number of bytes written via I2S
   size_t i2s_bytes_write = 0;

   for(int i=0;i<n;i++){
    // put the current sample in fy
    fy=*data16;
    
    //making this value larger will decrease the volume(Very simple DSP!). 
    fy/=1;
    
    // write data to I2S buffer 
    i2s_write(I2S_NUM_0, &fy, 2, &i2s_bytes_write,  10 );
    
    //move to next memory address housing 16 bit data 
    data16++;
   }
}

void setup() {

  // i2s configuration
  static const i2s_config_t i2s_config = {
    .mode = static_cast<i2s_mode_t>(I2S_MODE_MASTER | I2S_MODE_TX),
    .sample_rate = 44100,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = static_cast<i2s_comm_format_t>(I2S_COMM_FORMAT_I2S|I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 1000,
    .use_apll = false,
    .tx_desc_auto_clear = true
  };
  
  // i2s pinout
  static const i2s_pin_config_t pin_config = {
    .bck_io_num = 26,//26
    .ws_io_num = 27,
    .data_out_num = 25, //
    .data_in_num = I2S_PIN_NO_CHANGE
  };
  
  // now configure i2s with constructed pinout and config
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  i2s_set_pin(I2S_NUM_0, &pin_config);
  i2s_set_clk(I2S_NUM_0, 44100, I2S_BITS_PER_SAMPLE_16BIT, I2S_CHANNEL_STEREO);
  i2s_set_sample_rates(I2S_NUM_0, 44100);
  
 // set up bluetooth classic via bluedroid
  btStart();
  esp_bluedroid_init();
  esp_bluedroid_enable();

 
  // set up device name
  const char *dev_name = "ESP_SPEAKER";
  esp_bt_dev_set_device_name(dev_name);

  // initialize A2DP sink and set the data callback(A2DP is bluetooth audio)
  esp_a2d_sink_register_data_callback(bt_data_cb);
  esp_a2d_sink_init();
  
  // set discoverable and connectable mode, wait to be connected
  //esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE); 
  esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);

}
void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
}`

Not my implementation, found online.
A huge minus of BT audio esp32 is the amount of memory it takes. Internet radio + BT audio requires ~ 1.9 MB flash esp32.

closed 2023-05-15 by stale[bot]