schreibfaul1/ESP32-audioI2S #479: Slowly pulsing noise when playing mp3 (but not aac)

redfast00, 2023-02-07:
I'm trying to play an MP3 file on SPIFFS. The code plays the file, but there's slow, pulsing noise. This is both when playing from internet streams, as when playing from SPIFFS. When I use an AAC stream, I don't have this problem.

I'm using the Lolin S2 mini board and had to change the pins:

```
#define I2S_DOUT      9
#define I2S_BCLK      7
#define I2S_LRC       6
```

I'd prefer to use MP3, since this seems to compress better and I'd be able to fit more audio on the board.

redfast00, 2023-04-15:
@HelioDB do you also have this issue?

redfast00, 2023-05-02:
@HelloDB do you also have this issue?

schreibfaul1, 2023-05-02:
The ESP32-S2 only has one core, that won't work. Please use an ESP32.

redfast00, 2023-05-02:
Oh, okay, thanks for the information. Would you accept a PR that adds this to the readme?

closed 2024-03-13 by github-actions[bot]