schreibfaul1/ESP32-audioI2S #480: Can't play chaosradio.de mp3 URL

probonopd, 2023-02-09:
Thanks @schreibfaul1 for this very useful library.

Using https://github.com/probonopd/esp32-i2s-webradio/, can't play the mp3 URL below on a [ESP32-PICO-KIT V4](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/get-started-pico-kit.html) with i2s DAC.

```
audio_info: Connect to new host: "https://chaosradio.de/podlove/file/2054/s/webplayer/cr274-reverse-engineering.mp3"
audio_info: buffers freed, free Heap: 152912 bytes
audio_info: SSL has been established in 1299 ms, free Heap: 110460 bytes
Requested to play: https://chaosradio.de/podlove/file/2054/s/webplayer/cr274-reverse-engineering.mp3
audio_info: chunked data transfer
audio_info: redirect to new extension at existing host "https://chaosradio.de/archive/cr274-reverse-engineering.mp3?ptm_source=webplayer&ptm_context=&ptm_request=c2abb9d65ab4&ptm_file=cr274-reverse-engineering.mp3"
audio_info: new request: "https://chaosradio.de/archive/cr274-reverse-engineering.mp3?ptm_source=webplayer&ptm_context=&ptm_request=c2abb9d65ab4&ptm_file=cr274-reverse-engineering.mp3"
audio_info: Filename is 
audio_info: MP3Decoder has been initialized, free Heap: 75176 bytes
lasthost    https://chaosradio.de/podlove/file/2054/s/webplayer/cr274-reverse-engineering.mp3
audio_info: stream ready
buffer filled in 31276 ms
audio_info: ID3 framesSize: 307808
audio_info: ID3 version: 2.3
audio_info: ID3 normal frames
id3data     Title: Reverse Engineering
audio_info: slow stream, dropouts are possible
audio_info: Audio-Length: 100222431
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: slow stream, dropouts are possible
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
audio_info: syncword not found
```

The exact combinations of library versions I am using can be seen [here](https://github.com/probonopd/esp32-i2s-webradio/blob/main/platformio.ini).

probonopd, 2023-02-09:
On the other hand, https://download.deutschlandfunk.de/file/dradio/2023/02/08/psychosoziale_folgen_der_corona_pandemie_dlf_20230208_2346_1a9334b5.mp3 plays perfectly fine.

schreibfaul1, 2023-02-09:
The file contains a large image (>200KB) which is skipped.
This was a timing issue and resulted in an incorrect jump address. Thank you for your contribution.

probonopd, 2023-02-09:
I confirm that it is working for me. Thank you very much @schreibfaul1 :+1:

closed 2023-02-09 by probonopd