schreibfaul1/ESP32-audioI2S #481: Can't play chaosradio.de m4a URL

probonopd, 2023-02-09:
Thanks @schreibfaul1 for this very useful library.

Using https://github.com/probonopd/esp32-i2s-webradio/, can't play the mp3 URL below on a [ESP32-PICO-KIT V4](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/get-started-pico-kit.html) with i2s DAC.

```
audio_info: Connect to new host: "https://chaosradio.de/podlove/file/2053/s/webplayer/cr274-reverse-engineering.m4a"
audio_info: buffers freed, free Heap: 162312 bytes
audio_info: SSL has been established in 1870 ms, free Heap: 118924 bytes
Requested to play: https://chaosradio.de/podlove/file/2053/s/webplayer/cr274-reverse-engineering.m4a
audio_info: chunked data transfer
audio_info: redirect to new extension at existing host "https://chaosradio.de/archive/cr274-reverse-engineering.m4a?ptm_source=webplayer&ptm_context=&ptm_request=3b957961c524&ptm_file=cr274-reverse-engineering.m4a"
audio_info: new request: "https://chaosradio.de/archive/cr274-reverse-engineering.m4a?ptm_source=webplayer&ptm_context=&ptm_request=3b957961c524&ptm_file=cr274-reverse-engineering.m4a"
audio_info: Filename is 
audio_info: AACDecoder has been initialized, free Heap: 93372 bytes
lasthost    https://chaosradio.de/podlove/file/2053/s/webplayer/cr274-reverse-engineering.m4a
audio_info: stream ready
                        buffer filled in 6469 ms
[  6471][E][Audio.cpp:1850] read_M4A_Header(): subtype 'MA4 ', 'isom' or 'mp42' expected, but found 'M4V  '
[  6476][E][Audio.cpp:1195] readAudioHeader(): Processing stopped due to invalid audio header
```

The exact combinations of library versions I am using can be seen [here](https://github.com/probonopd/esp32-i2s-webradio/blob/main/platformio.ini).

schreibfaul1, 2023-02-09:
it's not just audio, I can't say whether the 3 other streams can simply be hidden, I'll put the problem on the waiting list for now

probonopd, 2023-04-14:
.

probonopd, 2023-06-18:
c

probonopd, 2024-03-15:
That bot is annoying. Closing valid tickets just due to inactivity practically guarantees further inactivity, and makes it hard for new contributors to see what could be worked on.