schreibfaul1/ESP32-audioI2S #493: DLNA Example produce error while compiling

pdurys, 2023-02-14:
With latest version 3.0.0 there is DLNA example, which #include websrv.h header.
This is missing in my setup and there is no such package available
```
lib/ESP32-audioI2S/examples/DLNA/src/main.cpp:3:10: fatal error: websrv.h: No such file or directory

****************************************************************
* Looking for websrv.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:websrv.h"
* Web  > https://registry.platformio.org/search?q=header:websrv.h
*
****************************************************************

 #include "websrv.h"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32-s3-devkitc-1\src\lib\ESP32-audioI2S\examples\DLNA\src\main.cpp.o] Error 1
============================================== [FAILED] Took 14.29 seconds ==============================================
```

I have downloaded library zip, extracted into Project lib folder and updated lib_deps in platformio.ini

I have simply deleted DLNA folder to pass over this error.
I have also error with another example time_announcement.
```
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
c:/users/xxxxx/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/src/main.cpp.o:C:\Users\xxxxx\Documents\PlatformIO\Projects\Radio/src/main.cpp:27: multiple definition of `audio'; .pio/build/esp32-s3-devkitc-1/src/lib/ESP32-audioI2S/examples/time_announcement/main.cpp.o:C:\Users\xxxxx\Documents\PlatformIO\Projects\Radio/lib/ESP32-audioI2S/examples/time_announcement/main.cpp:21: first defined here
c:/users/xxxxx/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/src/main.cpp.o: in function `loop()':       
C:\Users\xxxxx\Documents\PlatformIO\Projects\Radio/src/main.cpp:235: multiple definition of `loop()'; .pio/build/esp32-s3-devkitc-1/src/lib/ESP32-audioI2S/examples/time_announcement/main.cpp.o:C:\Users\xxxxx\Documents\PlatformIO\Projects\Radio/lib/ESP32-audioI2S/examples/time_announcement/main.cpp:71: first defined here
c:/users/xxxxx/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-s3-devkitc-1/src/src/main.cpp.o: in function `setup()':      
C:\Users\xxxxx\Documents\PlatformIO\Projects\Radio/src/main.cpp:91: multiple definition of `setup()'; .pio/build/esp32-s3-devkitc-1/src/lib/ESP32-audioI2S/examples/time_announcement/main.cpp.o:C:\Users\xxxxx\Documents\PlatformIO\Projects\Radio/lib/ESP32-audioI2S/examples/time_announcement/main.cpp:62: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1
============================================= [FAILED] Took 146.29 seconds =============================================

```

It looks like only those two examples are having PIO structures not simple ino file

schreibfaul1, 2023-02-14:
Yes, this is easier to use as a complete project than copying individual files into an IDF. There are also some dependencies that are easier to resolve this way.

pdurys, 2023-02-14:
Fair enough. Maybe add some text in those folders to inform about that side effects.

closed 2023-02-14 by pdurys