schreibfaul1/ESP32-audioI2S #495: Hostaddress is too long

aidam38, 2023-02-15:
Hello,

Firstly, thank you so much for adding the OPUS support! <3 

The youtube links I'm trying to play are longer than 512 characters, would it be possible to increase the hostaddress limit here https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L401?

schreibfaul1, 2023-02-15:
So I have two questions:
1) how long are youtube links?
2) do you use youtube-dl or does it work differently?

aidam38, 2023-02-15:
1. The expanded links are about 800 to 900. Here's an example:
```
https://rr5---sn-jxopj-n5oe.googlevideo.com/videoplayback?expire=1676446608&ei=MDfsY7i7CdjSkgbuzIzQAQ&ip=131.215.220.164&id=o-ACq8HwPcDMJ39TIXHG5Zq5HRSLQQdRg7Dk4Y79wJuQrp&itag=249&source=youtube&requiressl=yes&mh=dB&mm=31%2C29&mn=sn-jxopj-n5oe%2Csn-a5mekn6r&ms=au%2Crdu&mv=m&mvi=5&pl=16&initcwndbps=1342500&vprv=1&mime=audio%2Fwebm&ns=SKeNMz1xHPl75erOrbl0dd4L&gir=yes&clen=4051644&dur=642.961&lmt=1674300524681678&mt=1676424532&fvip=3&keepalive=yes&fexp=24007246&c=WEB&txp=3318224&n=5GCB7P_JHI4m1g&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cvprv%2Cmime%2Cns%2Cgir%2Cclen%2Cdur%2Clmt&sig=AOq0QJ8wRAIgfrgSbV22CHR-zG4InvD8xhovg7ilXmwldobonUL4ExQCIHUn3zupHnX5Q2TEBRVLpPv_5lw9w5QQlMLvoT2cJIp1&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRAIgPO3fx2de9uounQze2b5wqUzzwMm_udwVI9N_2AgI8NwCICMTEbhNc38Jb28bwVCG7ZpyO1M9-zJufT-9KfQeeezs
```

I played with the code a little and the expected length of the extension after url encoding is about 1050 (`extLen` here https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L435), which is notable more than 1024.

I tried just stupidly removing the check at line https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/src/Audio.cpp#L401 but I was getting stack overflow errors I think? Might be the Arduino setup stack running out?

2. Yes, I use youtube-dl. There are many ways to use youtube-dl, I wrote out how to use it from a node.js script here https://github.com/schreibfaul1/ESP32-Opus-Player/issues/1, but you can also get it from just the command line.

aidam38, 2023-02-23:
The url gets parsed correctly after the latest commit.

The new issue is that the code currently doesn't support the `audio/webm codecs=opus` container, which is distinct from `ogg`, but still uses the `opus` encoding under the hood. Letting `audio/webm` be parsed same as `audio/ogg` didn't work.

schreibfaul1, 2023-02-23:
I didn't have audio/webm as the content type yet. That seems to be a multimedia codec. You need a video player to play this?

aidam38, 2023-02-23:
It's not a codec, it's a container, and yes, it can also wrap video, but it can also wrap only audio. Idk why youtube uses it even for audioonly endpoints, maybe it's just convenient. You can read more about it here: https://developer.mozilla.org/en-US/docs/Web/Media/Formats/Audio_codecs .

schreibfaul1, 2023-02-24:
Thanks for the clarification. If webm is a wrapper similar to ogg, how difficult is it to extract the audio stream?

aidam38, 2023-02-24:
Unfortunately, I'm not an expert here. After a quick Google search, it seems like there isn't a webm container decoder in the Xiph.org libraries, but I found this repository from the WebM project which I think could serve as a basis for the code. https://github.com/webmproject/libvpx/

closed 2023-06-18 by stale[bot]