schreibfaul1/ESP32-audioI2S #498: Cannot play files from hosts that require User-Agent header

probonopd, 2023-02-18:
In https://github.com/schreibfaul1/ESP32-audioI2S/issues/363, the User-Agent header was commented out.

Unfortunately this breaks playing from hosts that require a User-Agent header.

Could it please be made configurable by allowing to optionally pass the desired User-Agent string to `Audio::connecttohost`, similar to `user` and `pwd`?

schreibfaul1, 2023-02-19:
How could this work without affecting other hosts and without adding more parameters?
Do the hosts that reject the connection come up with a unique error number? Then the connection could be re-initiated with a 'User Agent'.

probonopd, 2023-02-20:
Maybe have a `Audio::connecttohostwithuseragent`?

closed 2023-06-18 by stale[bot]

probonopd, 2023-06-18:
This bot is annoying. It prevents things from being solved by just closing them.