schreibfaul1/ESP32-audioI2S #499: ESP crashes in certain programme constellations with MP4 files

anp59, 2023-02-21:
Hi, during a test I noticed that the playback of local m4a files (AAC) leads to a programme crash.
Strangely, this crash does not occur if the audio event audio_info is implemented. If this function is commented out in main.cpp, the programme crashes again.
This bug has been tested with the current software and the standard example, both with and without SdFat support. A test file is attached.
The problem has already been reported with issue #439, but has not been dealt with yet.
Greetings and thanks in advance.
Andreas

[test_aac.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/10798718/test_aac.zip)

schreibfaul1, 2023-02-22:
Can someone reading this please test it on their system. The file runs with and without PSRAM. Anyone have similar errors?

````
ESP32 Chip: ESP32-D0WDQ6
Arduino Version: 2.0.6
ESP-IDF Version: 4.4.3
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
info        PSRAM not found, inputBufferSize: 6399 bytes
info        buffers freed, free Heap: 154632 bytes
info        Reading file: "/m4a/test_aac.m4a"
info        AACDecoder has been initialized, free Heap: 129640 bytes
id3data     Title: Ii.   Danses Caract Ristiques: Danse Russe (Trepak)
id3data     Artist: Sir Georg Solti
id3data     Album: The Tchaikovsky Album- Chicago Symphony Orchestra- Solti
id3data     Encoder: iTunes v4.0.1, QuickTime 6.3
id3data     Composer: Peter Ilyich Tchaikovsky
id3data     BPM: 
id3data     Track Number: 
id3data     Year: 1997
id3data     Compile: 
id3data     Types of: Classical
info        AudioType: MPEG4 / Audio
info        max bitrate: 128000
info        avr bitrate: 128000
info        AudioObjectType: AAC Low Complexity
info        Sampling Frequency: 44100
info        Channel Configurations: front-left, front-right
info        AAC FrameLength: 1024 bytes
info        ch; 2, bps: 16, sr: 44100
info        Audio-Length: 1080855
info        Content-Length: 0
info        stream ready
info        syncword found at pos 0
info        Channels: 2
info        SampleRate: 44100
info        BitsPerSample: 16
info        BitRate: 47545
info        AAC HeaderFormat: RAW
info        VBR recognized, audioFileDuration is estimated
info        Closing audio file
info        End of file "test_aac.m4a"
eof_mp3     test_aac.m4a
````

anp59, 2023-02-22:
Hi,
Enclosed is the log of the crash (InstrFetchProhibited). 
This occurs when the optional audio event function `void audio_info(const char *info)` is not defined in the programme.
```

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060730  A0      : 0x800dbdd4  A1      : 0x3ffb1f70
A2      : 0x3ffc40ac  A3      : 0x3ffc4f18  A4      : 0x3ffb1fb0  A5      : 0x00000003  
A6      : 0x00000000  A7      : 0x006e776f  A8      : 0x800dacf4  A9      : 0x3ffb1eb0
A10     : 0x3ffbb098  A11     : 0x3f4030f4  A12     : 0x000000a0  A13     : 0x00000000  
A14     : 0x46494441  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4008a6e5  LEND    : 0x4008a6f5  LCOUNT  : 0xfffffffe  

Backtrace: 0xfffffffd:0x3ffb1f70 0x400dbdd1:0x3ffb2270 0x400dd7ab:0x3ffb2290 0x400dd877:0x3ffb2510 0x400debad:0x3ffb2530 0x400e223f:0x3ffb27c0 0x400d3262:0x3ffb27e0 0x400f2ded:0x3ffb2820

  #0  0xfffffffd:0x3ffb1f70 in ?? ??:0
  #1  0x400dbdd1:0x3ffb2270 in Audio::setDecoderItems() at lib/ESP32-audioI2S/src/Audio.cpp:3885
  #2  0x400dd7ab:0x3ffb2290 in Audio::sendBytes(unsigned char*, unsigned int) at lib/ESP32-audioI2S/src/Audio.cpp:3969
  #3  0x400dd877:0x3ffb2510 in Audio::playAudioData() at lib/ESP32-audioI2S/src/Audio.cpp:3245
  #4  0x400debad:0x3ffb2530 in Audio::processLocalFile() at lib/ESP32-audioI2S/src/Audio.cpp:2860 (discriminator 1)
  #5  0x400e223f:0x3ffb27c0 in Audio::loop() at lib/ESP32-audioI2S/src/Audio.cpp:2265
  #6  0x400d3262:0x3ffb27e0 in loop() at src/main.cpp:188
  #7  0x400f2ded:0x3ffb2820 in loopTask(void*) at C:/Users/ANP/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50 

```
The troubleshooting revealed that in `void Audio::showCodecParams()` the function pointer to audio_info is used unchecked. 
My solution is to additionally check the validity of these pointer in line 3787 of audio.cpp:

`if((m_codec == CODEC_AAC || m_codec == CODEC_M4A) && audio_info) { ...`

With this change, the error no longer occurs. :-)

schreibfaul1, 2023-02-23:
Thanks, the error only occurs when something is sent to audio_info() and the audio_info() routine doesn't exist. That should be fixed now.

anp59, 2023-02-23:
Hi Wolle, 

wouldn't it be better to use m_chbuf in the macro AUDIO_INFO and put the query „if ( audio_info )“ at the beginning, i.e. before sprintf?  Currently the macro uses almost 600 bytes of the stack.

Many greetings 

Andreas

schreibfaul1, 2023-02-23:
Hi Andreas,
you are right, there is still room for improvement. I moved the memory to PSRAM, and when that doesn't work, to the heap. Now there is a little more space in the stack :-))

Many greetings
Wolle

closed 2023-06-18 by stale[bot]