schreibfaul1/ESP32-audioI2S #501: Equalizer-Einstellung.

spikepavel, 2023-02-23:
Gute Nacht! Mein Name ist Paul. Ich komme aus einer kleinen Stadt namens Elista. Wir haben fast 1000 Leute, die Ihre ESP32-audioI2S-Bibliothek nutzen, sie ist sehr gut und wird ständig von Ihnen aktualisiert. Aus diesem Grund sehr beliebt. Ich lade es herunter, ändere es ein wenig und verteile es dann an andere Benutzer weiter. Ich habe einen Vorschlag an Sie, Audio vom Mikrofon auf die SD-Karte aufzunehmen. Umso einfacher wird es mit den Mitteln Ihrer Bibliothek, ein paar Zeilen zum Hinzufügen von Mikrofoneingangskontakten, ein paar Zeilen zum Puffer, einen Befehl zum Schreiben und Stoppen und dann speichern. Es wird definitiv nicht schlimmer, aber ich denke, es wird die Popularität erhöhen.

Noch etwas ist mit der Equalizer-Einstellung nicht ganz klar, wenn Sie Einstellungen über 0,0,0 einstellen, wird der Ton leiser. Ich habe viele Male versucht, an Ihre E-Mail zu schreiben, aber der Brief kommt nicht an.
 
Danke für die geleistete Arbeit. Gesundheit, Glück und alles Gute!

schreibfaul1, 2023-02-23:
Hallo @spikepavel,
erst einmal vielen Dank für die netten Worte.
Der Equalizer kann anders als die passiven Netzwerke der analogen Welt das Signal verstärken. +6dB bedeutet eine Verdoppelung des Signals. In den älteren Versionen hatte ich das Signal  vor dem Equilizer immer auf 50% reduziert. Das bedeutete aber auch, dass das Signal bei 0,0,0 nur 0.5 * max Lautstärke erreicht hat. Jetzt wird bei einer möglichen Verstärkung durch den Equilizer das Gesamtsignal um diesen Wert reduziert. Die "Samples" bewegen sich im ``int16_t`` Bereich. Eine geringe Überschreitung nach oben oder unten führt zu extremen Verzerrungen.
Dafür lassen sich die Lautstärkelevel jetzt feinstufiger einstellen. Standard ist immer noch 0...21, das lässt sich jetzt bei Bedarf bis 255 erhöhen ``setVolumeSteps()``
Demnächst kommt ein VORBIS(ogg) Dekoder hinzu, OPUS ist seit Januar an Start. Damit erreiche ich den Funktionsumfang eines  VS1053B.
Eine Mikrofonfunktion wäre manges Enkoder auf wav beschränkt. 
beste Grüße
Wolle

spikepavel, 2023-02-24:
Guten Tag. Ich bin sehr froh, dass Sie mir geantwortet haben. Vielen Dank für Ihre Arbeit! Ich schließe das Thema.

closed 2023-02-24 by spikepavel