schreibfaul1/ESP32-audioI2S #504: ID3 callback for GUI

moononournation, 2023-03-04:
I would like to build a ESP32 music play with a touch screen display. The GUI will show the music ID3 information. But I found the `audio_id3data()` concatenated the tag name, colon, space and then the real ID3 data. I need to break it down again for displaying. Is it possibly change the callback that can use the ID3 data easily?
And also I found the lyrics tag only can return around first 200 characters, it is not enough for most song.

spikepavel, 2023-03-04:
Hello. You can simply translate the received information into a string, and then parse it immediately before output. The beginning will be the same, you can cut off or create your own team that will already output.

schreibfaul1, 2023-03-04:
Hi, @moononournation,
you are right, the metadata is war (ID3v1) limited to ~128bytes. ID3v2 does not have this limitation. For the users who use an ESP32 without PSRAM, I cannot reserve much memory for the metadata. I doubled the internal buffer. Unicode now reaches ~244 characters, without Unicode about twice that is possible.

Thanks @spikepavel, that's how I would do it,
or if the 'Year: ' in 'Year: 2003' bothers you, simply move the pointer further to the right.
````
void audio_id3data(const char *info){  //id3 metadata
    if(strncmp(info, "Year: ", 6) == 0) Serial.println(info + 6);
}
````

moononournation, 2023-03-04:
I know it can handle in this way, but I mean it introduces too much overhead. You need sprintf to output this string, then user need a nested if condition of strcmp check it is which tag, then trash out the header and then use the actual ID3 data it have at the beginning of internal private scope. It use up too much memory and processing power for each MP3 start.

closed 2023-06-18 by stale[bot]