schreibfaul1/ESP32-audioI2S #505: Infinite loop with empty or corrupt short local file

anp59, 2023-03-05:
Hi, a local file (MP3 or other music files) with a file size of 0 or a few bytes causes an infinite loop in the code of ProcessLocalFile(). A recognition of the end of the file (eof event) or better the triggering of an additional error event would be desirable.

schreibfaul1, 2023-03-07:
I created a short file.
[5ms.mp3.txt](https://github.com/schreibfaul1/ESP32-audioI2S/files/10907390/5ms.mp3.txt)
That's a 1KHz sine wave, 5ms long, that's 5 oscillations.
[image]
There are no problems with that.

anp59, 2023-03-07:
I noticed the infinite loop with an empty .opus file of a playlist. A short file with a size of 5 bytes (no MP3 format, but ending .mp3) showed the same behaviour. Normally, such files will not occur. For a playlist, however, it would be important if at least one eof event were triggered here as well.

schreibfaul1, 2023-03-07:
Hi @anp59,
I could understand that now. I created a file using the text editor. If the audioheader cannot be read, this leads to an infinite loop. This is now terminated by a timeout.

closed 2023-06-18 by stale[bot]