schreibfaul1/ESP32-audioI2S #512: Support for Tonie Audio Files (Opus file with custom file header)

SciLor, 2023-03-30:
I would like to play Tonie Audio Files short TAFs with your library, TAFs are just OPUS files with a prepended custom 4k header.
So it would be enough to just skip 4k of the file to use it as valid input data. 

In addition, it would be nice to create tracks by a cue file and a single audio file. TAFs contain a chapter list which would do just that. Splitting a single audio file in multiple tracks. The TAF contains a list of ogg page numbers for that.

Questions:
- Is it possible to play the TAFs by skipping the first 4k of the file?
- Is it possible to split a single audio file in multiple tracks?

If not, what would you think are the best points to extend your lib?

https://github.com/toniebox-reverse-engineering/toniebox/wiki/Audio-file-format

schreibfaul1, 2023-03-31:
TAP appears to be a proprietary format that does not conform to the OPUS or OGG standard.

SciLor, 2023-03-31:
The TAFs seem to be a valid OGG container with OPUS pages if you cut the first 0x1000.
And in the OGG container each OPUS page is padded to 0x1000  FFMPEG takes it as valid OPUS file.

Yesterday I have renamed a TAF to .opus, so audioI2S detects the codec. In addition, I have cut the first 0x1000 bytes of a TAF.

PS: I will add a log of both testcases

closed 2024-02-29 by github-actions[bot]