schreibfaul1/ESP32-audioI2S #516: library working on esp32 version 1.0.6, but not working on esp32 version 2.0.0 and above?

michaelknight999, 2023-04-06:
hi,
not sure if I'm doing something wrong or it could be a bug.

migrating some working code from old desktop to a new one, I found problems running an existing sketch in a esp32 board in arduino IDE. (arduino IDE v1 and v2, same results in both cases)

in the old desktop, if I upload the sketch to the esp32 board, everything runs correctly. compilation is successful, upload is successful and mp3 are played successfully.
howerver, doing the same from new desktop, is not the same. compilation is successful, upload is successful but mp3 are not played. all rest of the code runs well, but there is no sound (and no errors are shown either).

after hours investigating and comparing both environments, I found that the difference between both systems is that in old desktop I had installed esp32 boards version 1.0.6; and in the new desktop the last version of esp32 boards 2.0.7 was installed.

after downgrade the esp32 board version to 1.0.6 in new desktop, everything run correctly again and the sound was back.

I tested also with esp32 board versiones 2.0.0, 2.0.1 and 2.0.5 and there were no sound either.

so, is the library only working in esp32 boards version below 2.0.0? or am I doing some wrong or missing any step to make it work in last 2 versions?

if you need additional info, just let me know.

thanks!

schreibfaul1, 2023-04-06:
With the change of the Arduino version from V1.0.x to V2.0.x, something has changed in the initialization of the I2S system. However, the V2 is recognized and I2S is initialized accordingly. I use V2.0.6 without problems and many other users too.
Maybe the IDF version is still very old?

````c++
    uint32_t idf = ESP_IDF_VERSION_PATCH + ESP_IDF_VERSION_MINOR *10 + ESP_IDF_VERSION_MAJOR * 100;
    uint32_t ard = ESP_ARDUINO_VERSION_PATCH + ESP_ARDUINO_VERSION_MINOR * 10 + ESP_ARDUINO_VERSION_MAJOR * 100;
    log_i("ESP32 IDF Version %d.%d.%d", idf/100%10, idf/10%10 , idf%10);
    log_i("ESP32 Arduino Version %d.%d.%d", ard/100%10, ard/10%10 , ard%10);
````
shows me:
[   492][I][main.cpp:47] setup(): ESP32 IDF Version 4.4.3
[   492][I][main.cpp:48] setup(): ESP32 Arduino Version 2.0.6

michaelknight999, 2023-04-08:
I checked the idf version from arduino (as far as I know arduino works over idf api, correct?) using next command (Serial.println(esp_get_idf_version());), and I get the next results:

- old desktop using arduino esp32 board version 1.0.6 working successfully:
  - idf version: v3.3.5-1-g85c43024c

- new desktop, testing different versions of arduino esp32 boards:
  - version 1.0.6, works correctly:
idf version: v3.3.5-1-g85c43024c

  - version 2.0.6, no errors but no sound:
idf version: v4.4.3

  - vesrion 2.0.7, no errors but no sound:
idf version: v4.4.4

does it make any sense?

thanks!

michaelknight999, 2023-04-15:
I could find the time to create a little test project to try find where is the problem, and after testing the internet radio example I can confirm the library works fine with arduino esp32 board versions 2.0.6 and 2.0.7.

so it looks like I have some kind of error or not correct configuration in the old project (I suspect now about the use of some gpio pins for SD card reader in the wrong way or that changed the use in the new version). I will try to find the problem and and a last comment just in case it is useful for other people with the same problem.

but I think we can close the ticket as it is not a library bug

thanks!

closed 2023-04-15 by michaelknight999