schreibfaul1/ESP32-audioI2S #517: A new PCM5102 board has been failing the sound of certain web radio stations.

renatoianhez, 2023-04-06:
I have built [this radio](https://github.com/renatoianhez/Lolin32-radio) which works very well with its excellent library and which I have gifted to my friends. I used [this PCM5102](https://pt.aliexpress.com/item/1005003635257502.html) board with no problems. Now, I got [this PCM5102](https://pt.aliexpress.com/item/32828177686.html) that I used with this wiring diagram below. Some radios are behaving strangely, with low and faulty sound. The questions: a) What is the difference between one plate and the other? b) What characteristic of the radio stations below causes them to fail? c) Is there something in the library configuration that I need to change?

Thanks!!

[image]
http://cloud2.cdnseguro.com:23538/; // Kiss FM
https://24083.live.streamtheworld.com/JBFMAAC.aac // JB FM
http://20323.live.streamtheworld.com/RADIO_ALVORADAAAC_SC // Alvorada
https://s16.maxcast.com.br:8028/live // Estacao Reggae
http://euroticast5.euroti.com.br:8070/; // Axe BA
http://50.3.30.111:8042/stream // Seresta ao Luar
http://media-ice.musicradio.com/RadioXLondon // XFM
http://ice7.securenetsystems.net/WROD // WROD
http://8642.brasilstream.com.br/stream?identifica=9725815160273912000&rand=9201410188400340000 //UFES

schreibfaul1, 2023-04-06:
Maybe someone reading along here and owning this board can share their experiences here.
This could probably be a timing problem. The bit rate is a criterion here. Line lengths shouldn't cause any problems with the low transmission rates on the I2S.

probonopd, 2023-04-15:
Using https://github.com/probonopd/esp32-i2s-webradio/ (which uses this library) I can play all streams without issues on a ESP32CAM module (with PSRAM) and a MAX98357A (3 Watt amplifier with DAC). Some have better quality than others, but all seem to play without issues. Maybe you should try a module with PSRAM.

The last stream (brasilstream.com.br) gives `HTTP/1.1 502 Bad Gateway`, possibly it cannot be played in my part of the world.

renatoianhez, 2023-04-15:
> Using https://github.com/probonopd/esp32-i2s-webradio/ (which uses this library) I can play all streams without issues on a ESP32CAM module (with PSRAM) and a MAX98357A (3 Watt amplifier with DAC). [...]

Oh, thanks! I have some of this module... I'll test! I use PSRAM via piggybacking. [See](https://github.com/renatoianhez/Lolin32-radio).

renatoianhez, 2023-04-15:
I also used, it seems to me, another version of Lolin32 Lite. I noticed the difference in the memory and ESP32 chips used.
[image]

renatoianhez, 2023-04-15:
I had a lot of work to select new stations that didn't give any problems! Now everything playing how I want.

renatoianhez, 2023-05-27:
The problem is the new Lolin32 Lite module! I believe it is the different memory chip.

closed 2024-02-28 by github-actions[bot]

01GOD, 2024-09-03:
> I have built [this radio](https://github.com/renatoianhez/Lolin32-radio) which works very well with its excellent library and which I have gifted to my friends. I used [this [...]

Hi! Ever got that board functioning with any Arduino IDE compatible examples? If did, can post a link to the code that did function with that PCM5102 DAC? 

Thanks in advance!

01GOD, 2024-09-03:
> I have built [this radio](https://github.com/renatoianhez/Lolin32-radio) which works very well with its excellent library and which I have gifted to my friends. I used [this [...]

By the way, FLT is apparently "filter selection", and connecting that to GND makes it select "normal latency" instead of "low latency". So maybe that was why.

schreibfaul1, 2024-09-03:
Was FLT the problem? I had something similar a long time ago. I think some sample rates didn't work. I can't remember what I did.