schreibfaul1/ESP32-audioI2S #521: Need to use this with GPRS connectivity

techiesms, 2023-04-18:
I tried this library with ESP32 and it's working fine. Now I want to play the audio via GPRS connectivity using SIM800 module. So do anyone know how can I make that project ?

schreibfaul1, 2023-04-18:
I looked at the data sheet of the SIM800, which seems to be a modem with a UART interface, which is controlled via AT commands. Developing something for this would take a lot of work. But maybe someone reading along here has already done something like this.

techiesms, 2023-04-19:
I can use ESP32 to play via I2S, that's not an issue, but I just want to use GPRS connectivity rather than WiFi Connectivity.
I want to download the audio file via GPRS rather than WiFi. So do you have any solution

schreibfaul1, 2023-04-19:
You need a routine that buffers the audio data via SIM800. If data is available, this can be done using a function
ret = SIM800.read(buff, size); passed to audioI2S. That would then work like an SD card.
For SD, SD:MMC, SPIFFS it works like this in Audio::processLocalFile().

techiesms, 2023-04-19:
is it possible for you to send a sample code for my reference purpose please ? I'm unable to get what you are saying that's why.

Or rather send me any link of document or blog that helps me with this

Thank You

schreibfaul1, 2023-04-19:
If the audio data is received via GPRS, then it certainly doesn't happen as a whole, but individual data packets are transmitted. If a data package has been received correctly, an 'ok' is sent to the remote station so that the next data package can be transmitted. (I assume that it is so) The received packet must now be somewhere. What you need is a read() function to write the data into the audioI2S inbuff. As exactly what SD or WiFiClient have. The read() function is periodically polled by audioI2S, that's all.

techiesms, 2023-04-19:
It's easy for you sir, but I'm unable to understand it. I'm not getting any function called SIM800.read() anywhere. 
It will be a great help if you can send a sample code of playing any random mp3 file over GPRS using ESP32 and SIM800

I'm really not able to understand but as you said, it seems an easier task for you. So kindly help me with this small thing

schreibfaul1, 2023-04-19:
Since I don't want to see through the SIM800 manuals, I asked ChatGPT, and ChatGPT says the module has a flash memory of 8 or 16MB depending on the manufacturer. Then there is certainly also the possibility of querying whether data is available or not. Your task is to write a read() function that works similarly to the SD card or the WiFi client. I can't help with that, I don't own such a module.

techiesms, 2023-04-19:
What I did till now it, I established GPRS connection and did GET request to get the data. I received the mp3 file data in HEX values and I stored that in SPIFFS memory of ESP32. But the I try to play that file from SPIFFS memory, I'm unable to play that. It shows the error syncword not found. 

So unable to play the audio.

when I did the same step for wav file, it play some noise and not able to get any information. 

now let me know if you can give me any solution

techiesms, 2023-04-19:
But I'm unable to store uint8_t data in SPIFFS. I need to store char datatype

schreibfaul1, 2023-04-19:
That's good too, the main thing is 8 bits. I remember that if the audio file comes from a web server, a response header containing values ​​such as contentlength and contenttype is sent at the beginning. The ResponseHeader ends with 
'\r\n\r\n' or '\n\n'
 and then comes the actual audio file.

techiesms, 2023-05-05:
I'm still not able to make it. If anyone reading this msg has made something similar please respond me with proper code or some documentation link. I wanted to make an MP3 player via GPRS connectivity

uoosef, 2023-11-28:
hello @techiesms how did you interface sim800 and esp32 with i2s? im doing something similar and i cant figure it out, because sim800 pcm can only be in master mode, can you send me a schematic and sample code ?

closed 2024-02-26 by github-actions[bot]