schreibfaul1/ESP32-audioI2S #523: No Matching function for call to 'ConnecttoFS(fs::SDFS&, String&)

The-Valveman, 2023-05-01:
Hi I am trying to an SD card based MP3 player to work i have tried various sketches douwnloaded  as i want an MP3 player to incorporate in a touch screen dab/FM receiver project that i am working on  i have the DAB/FM and internet radio functions working ok but i am struggling to get a MP3 player that reads MP3s from a local SD card to work i found the sketch below on github which would be perfect if i could just get it working  this sketch uses the ESP32- audioi2s library.   I have  tried an example by hard coding a track using the line 'audio.connecttoFS(SD,"/TRACK003.mp3.mp3"); ' and making sure the SD card contains an MP3 file with the name 'TRACK003.mp3 this works and the track plays perfectly.  But i need to be able get the player to read the SD card and allow the user to select which tracks to play and for it to play the tracks one after another until all tracks have been played.
when i try to compile the below sketch i get this error:-
 'Compilation error: no matching function for call to 'Audio::connecttoFS(fs::SDFS&, String&)'
If i comment out the line audio.connecttoFS(SD, filename); the code compiles but it doesn't play the files but it does read the SD card and lists the tracks on the card.
The code
**********************************************************************************************************************************

#include "Arduino.h"
//#include "WiFiMulti.h"
#include "Audio.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
 
//SD Card
#define SD_CS 22
#define SPI_MOSI 23
#define SPI_MISO 19
#define SPI_SCK 18
 
//Digital I/O used  //Makerfabs Audio V2.0
#define I2S_DOUT 27
#define I2S_BCLK 26
#define I2S_LRC 25
 
//SSD1306
#define MAKEPYTHON_ESP32_SDA 4
#define MAKEPYTHON_ESP32_SCL 5
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_RESET -1    // Reset pin # (or -1 if sharing Arduino reset pin)
 
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
 
//Button
const int Pin_vol_up = 39;
const int Pin_vol_down = 36;
const int Pin_mute = 35;
 
const int Pin_previous = 15;
const int Pin_pause = 33;
const int Pin_next = 2;
 
Audio audio;
 
struct Music_info
{
    String name;
    int length;
    int runtime;
    int volume;
    int status;
    int mute_volume;
} music_info = {"", 0, 0, 0, 0, 0};
 
String file_list[20];
int file_num = 0;
int file_index = 0;
 
void setup()
{
    //IO mode init
    pinMode(Pin_vol_up, INPUT_PULLUP);
    pinMode(Pin_vol_down, INPUT_PULLUP);
    pinMode(Pin_mute, INPUT_PULLUP);
    pinMode(Pin_previous, INPUT_PULLUP);
    pinMode(Pin_pause, INPUT_PULLUP);
    pinMode(Pin_next, INPUT_PULLUP);
[... 321 lines cut ...]
void audio_bitrate(const char *info)
{
    Serial.print("bitrate     ");
    Serial.println(info);
}
void audio_commercial(const char *info)
{ //duration in sec
    Serial.print("commercial  ");
    Serial.println(info);
}
void audio_icyurl(const char *info)
{ //homepage
    Serial.print("icyurl      ");
    Serial.println(info);
}
void audio_lasthost(const char *info)
{ //stream URL played
    Serial.print("lasthost    ");
    Serial.println(info);
}
void audio_eof_speech(const char *info)
{
    Serial.print("eof_speech  ");
    Serial.println(info);
}

schreibfaul1, 2023-05-03:
The call is: connecttoFS(fs::FS &fs, const char* path)
String is not const char*
use filename.c_str()

The-Valveman, 2023-05-03:
Thankyou @schreibfaul1 for your help that sorted it as it is now working. I am relatively new to programming this receiver project it certainly testing my somewhat limited programming skills. I have always been more comfortable handling a soldering iron and an oscilloscope than a keyboard lol.