schreibfaul1/ESP32-audioI2S #526: BLE connection causes SW_RST

PMKrol, 2023-05-03:
Hi!

I got into a problem. I'm trying to run audio.h on STM32-WROOM and also send some commands via BLE:

```
//01. WebServer based on https://github.com/smford/esp32-asyncwebserver-fileupload-example
//02. Audio based on //https://github.com/schreibfaul1/ESP32-audioI2S
//03. WiFi AP based on https://randomnerdtutorials.com/esp32-access-point-ap-web-server/
//04. BLE Android remote control: https://github.com/nkolban/ESP32_BLE_Arduino

//00. BLE as audio sink: TODO
//00. Hardware buttons & jack plug event: TODO
//00. remove authentication on web: TODO

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "8d910c6f-fe0e-4ea0-8ea1-55cce9a039d0"  //new
BLEUUID charUUID_cmd("90350001-c13c-4042-82d8-14387e6bd2fe");  //new
BLEUUID charUUID_p_p("90350002-c13c-4042-82d8-14387e6bd2fe");  //new
BLEUUID charUUID_vol("90350003-c13c-4042-82d8-14387e6bd2fe");  //new

BLEServer *pServer;
BLEService *pService;
BLECharacteristic *player_cmd;
BLECharacteristic *player_p_p;
BLECharacteristic *player_vol;
std::string cmd = {0x00};
std::string p_p = {0x00};
std::string vol = {0x0F};

#include "Arduino.h"
//#include "WiFiMulti.h"
#include "Audio.h"
#include "SPI.h"
//#include "SD.h"
#include "FS.h"

// Digital I/O used
#define SD_CS          5
#define SPI_MOSI      23
#define SPI_MISO      19
#define SPI_SCK       18
//#define I2S_DOUT      25
//#define I2S_BCLK      27
//#define I2S_LRC       26
#define I2S_DOUT      16  //DIN
#define I2S_BCLK      17  //BCLK
#define I2S_LRC       22   //WSEL

Audio audio;

#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <SD.h>
#include "webpages.h"

#define FIRMWARE_VERSION "FlacPlayer v0.0.4"

const String default_ssid = "RatamahattaTP";
const String default_wifipassword = "Hakunamatata";
const String default_httpuser = "admin";
const String default_httppassword = "admin";
const String wifiap_name = FIRMWARE_VERSION;
const String wifiap_password = "";
const int default_webserverporthttp = 80;

// configuration structure
[... 327 lines cut ...]

> assert failed: vQueueDelete queue.c:2131 (pxQueue)

```
Decoding stack results
0x40083915: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 408
0x400947b9: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c line 137
0x4009a279: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c line 85
0x40095bd6: vQueueDelete at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 2131
0x401615c1: osi_sem_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/semaphore.c line 75
0x401603fb: fixed_queue_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/fixed_queue.c line 88
0x4016044e: fixed_queue_new at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/fixed_queue.c line 67
0x4014e021: gatt_allocate_tcb_by_bdaddr at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_utils.c line 1087
0x4014c068: gatt_le_connect_cback at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/gatt/gatt_main.c line 490
0x4015a33d: l2cu_process_fixed_chnl_resp at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/l2cap/l2c_utils.c line 2873
0x40153cd6: l2cble_notify_le_connection at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/l2cap/l2c_ble.c line 278
0x4013e1d2: btm_ble_read_remote_features_complete at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/btm/btm_ble_gap.c line 4311
0x4014788c: btu_hcif_process_event at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/btu/btu_hcif.c line 2076
0x40147d12: btu_hci_msg_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/host/bluedroid/stack/btu/btu_task.c line 159
0x4016111b: osi_thread_run at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/bt/common/osi/thread.c line 165
```

Anyone can help me?

Best regards!

schreibfaul1, 2023-05-03:
No idea, the source code is quite long. Maybe two tasks access the same object, then you should set a message queue between them.

PMKrol, 2023-05-03:
I believe You can ignore everything except 

```
void setup() {
  pinMode(SD_CS, OUTPUT);      
  digitalWrite(SD_CS, HIGH);
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPI.setFrequency(1000000);
    
  Serial.begin(115200);
  Serial.print("Firmware: "); 
  Serial.println(FIRMWARE_VERSION);
  Serial.println("Booting ...");
  Serial.println("Mounting SD ...");
  //initSDCard();
  SD.begin(SD_CS);

  //TODO: złe wyniki
  Serial.print("SD Free: "); Serial.println(humanReadableSize((SD.totalBytes() - SD.usedBytes())));
  Serial.print("SD Used: "); Serial.println(humanReadableSize(SD.usedBytes()));
  Serial.print("SD Total: "); Serial.println(humanReadableSize(SD.totalBytes()));

  Serial.println(listFiles());
  Serial.println("Loading Configuration ...");

  config.ssid = default_ssid;
  config.wifipassword = default_wifipassword;
  config.httpuser = default_httpuser;
  config.httppassword = default_httppassword;
  config.webserverporthttp = default_webserverporthttp;
  config.wifiapname = wifiap_name;
  //config.wifiappassword = wifiap_password;

  initAudio();
  //initBLE();
  //initWeb();
  swichWiFiOrBLE();

}

void loop() {
  // reboot if we've told it to reboot
  if (shouldReboot) {
    rebootESP("Web Admin Initiated Reboot");
  }

  audio.loop();
  /*if(Serial.available()){ // put streamURL in serial monitor
    audio.stopSong();
    String r=Serial.readString(); r.trim();
    if(r.length()>5) audio.connecttohost(r.c_str());
    log_i("free heap=%i", ESP.getFreeHeap());
  }*/

  /*
  Serial.print(audio.getAudioCurrentTime());
  Serial.print("/");
  Serial.println(audio.getAudioFileDuration());
  */
  //Serial.println(listFiles());

}

...

void initAudio(){
  Serial.print("Starting audio... ");
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(12); // 0...21

  audio.connecttoFS(SD, "Flogging Molly - (2006) Whiskey on a Sunday.m4a ");
  Serial.println("done.");
}

...

void initBLE(){
  Serial.print("Starting BLE... ");
  
  BLEDevice::init(FIRMWARE_VERSION);
  pServer = BLEDevice::createServer();
  pService = pServer->createService(SERVICE_UUID);

  player_cmd = pService->createCharacteristic(charUUID_cmd, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE);
  player_p_p = pService->createCharacteristic(charUUID_p_p, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE);
  player_vol = pService->createCharacteristic(charUUID_vol, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE);

  player_cmd->setCallbacks(new MyBLECallbacks());
  player_p_p->setCallbacks(new MyBLECallbacks());
  player_vol->setCallbacks(new MyBLECallbacks());

  player_cmd->setValue(cmd);
  player_p_p->setValue(p_p);
  player_vol->setValue(vol);
  
  pService->start();

  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();

  Serial.println("done.");
}
```

closed 2023-05-04 by PMKrol