schreibfaul1/ESP32-audioI2S #531: unable to use the ib in a task ( for notifications, queues ..)

jalmince, 2023-05-10:
HI There!
Fisrt congrats for your job!

I would like to use the Audio in a freertos task that could receive the mp3 name to play via notification or queue.

All I could manage to do is sending the file name through a global var. the task doing the "Audio.loop()" checking
regurlarly for the global changes, and doing a "audio->connecttoFS(..) accordingly.

Could you by pls point me to a code template, or example, that does that, but with notifying or queueing ?
( in my countless hours of trying, I always get "assert failed: xTaskGenericNotify tasks.c:5545 (xTaskToNotify)"

PS: 
1) annoying thing is that every fail attempt forces me to re-flash the ESP32 with factory firmware, like if audio crash would have left the device in a wrong state? 
2) all I need is playing mp3 from sdcard. ( no streaming from internet ) Is there a stripped down version of the lib ?
may be an other lib you could point me to ?

kind regards,
Jean-Paul

schreibfaul1, 2023-05-10:
Hello Jean Paul,
normally audioI2S runs in the Arduino loop. For larger projects it may be necessary to run audioI2S in a separate task. You're right, then you need a message queue, possibly a semaphore for events.
Here is a simple example: https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/separate_audiotask

and here two complete projects:
https://github.com/biologist79/ESPuino
https://github.com/schreibfaul1/ESP32-MiniWebRadio

jalmince, 2023-05-10:
Thank you very much for replying !
I will study the examples you gave.
Best
Jpaul

jalmince, 2023-06-11:
Hello !

Thanks to your last reply and examples I am happily using the Audio lib in
a task.

However in my project I have also a task pinned on core 0 which displays
animated sprites on a tft display ( board is Lilygo T-display S3 )

The annoying behaviour is that the sprites freeze immediately, or after
some time, after the audio task has kicked-in, and comes back live again
after 1 or 2 mn !! ( and so on )

the audio task is started like so:  xTaskCreate( TaskPlay, "TaskPlay",
60000, NULL, 1, &pvCreatedTaskPlay);

the "display-sprites" task like so : xTaskCreatePinnedToCore( TaskEyes,
"TaskEyes", 10000, NULL, 0, &pvCreatedTaskEyes, 0);

- if I run the "display sprites" task with priority 1 ( like the audio )
the chips cycles on re-boots
- instead of streaming a web radio, Ive tried with an mp3 file on SDcard :
same thing, but the music plays kind of half tempo...
- you mention in the GIT that your lib only runs on esp32 with 2 cores. Is
it that another sub task kicks in on core 0 after some time ??

Unfortunately the pre-compiled freeRTOS version used in the arduino env
does not allow usage of vTaskList and I can't do much ...

I am really at sea!
If you have any hints or ideas where I could investigate you are welcome !

cheers
Jean-Paul

schreibfaul1, 2023-06-12:
Hello @jalmince ,
the display can run in the Arduino loop, no separate task is necessary for this. However, an audio task is required to avoid audio dropouts when drawing on the display. 60000 DWORDs for the stack is a lot, 5000 would be enough and you have more memory for your program.
The watchdog could have triggered during a reboot. The WDT must not remain unattended for too long.

closed 2024-02-25 by github-actions[bot]