schreibfaul1/ESP32-audioI2S #536: Audio is kinda broken while playing a radio

FightMan01, 2023-05-26:
I have an ESP32 and I'm using PlatformIO with framework arduino and espidf.
I tried with these 3 stations but I have the issue in all of them. 
- http://icast.connectmedia.hu/5016/live.mp3
- http://icast.connectmedia.hu/5001/live.mp3
- http://icast.connectmedia.hu/4738/mr2.mp3
Recording attached: https://diakoffice-my.sharepoint.com/:u:/g/personal/fightman01_sulid_hu/Ef7WDINd-pVLn6iZOGwHHbAB6KkrehfrBmE1wZObXwlrLQ?e=AIyc1P
Log:
```
audio_info: Connect to new host: "http://icast.connectmedia.hu/5016/live.mp3"
audio_info: PSRAM not found, inputBufferSize: 6399 bytes
audio_info: buffers freed, free Heap: 63316 bytes
audio_info: Connection has been established in 44 ms, free Heap: 62544 bytes
audio_info: chunked data transfer
audio_info: icy-name: Szombathely-TX
Szombathely-TX
audio_info: MP3Decoder has been initialized, free Heap: 32692 bytes
audio_info: stream ready
audio_info: syncword found at pos 0
audio_info: Channels: 2
audio_info: SampleRate: 44100
audio_info: BitsPerSample: 16
audio_info: BitRate: 192000
audio_info: StreamTitle=''
```
I think it is worth mentioning that I have set `WiFi.setSleep(false);` because without it I'm getting slow stream, dropouts are possible and a lot of stream reconnects. Also when I start the stream the first 1-2 second is fine then the audio just breaks.

schreibfaul1, 2023-05-26:
Hello,
Arduino is optimized for HW, not for max. WiFi speed. If you use Arduino + EspIDF, you can simply change some values ​​in menuconfig in the TCP area so that the streams mentioned can be played without interruptions. 
The precompiled Arduino is sufficient for many streams. Sometimes, probably depending on the size of the TCP packets, changes in the TCP settings are necessary.
Look here: https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput

FightMan01, 2023-05-27:
Thank you for the reply and for the help. 
It worked fine.

closed 2023-05-27 by FightMan01