schreibfaul1/ESP32-audioI2S #537: how use psram with esp32s3?

ddgit123, 2023-05-27:
hello,everybody:
I used esp32s3 to create an internet radio with this library, but it often prompts
"info slow stream, dropouts are possible"
Can I use PSRAM to improve it? How can I use PSRAM on esp32s3?
Thank you!

schreibfaul1, 2023-05-27:
Using PSRAM is a good idea. As far as possible, buffers are allocated in the PSRAM, which is what all decoders do. Unfortunately, this does not solve the problem with the low WiFi throughput. This requires settings in the menuconfig. https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput

dmiro, 2023-09-06:
@ddgit123  I have same issue. I have an esp32-wroom v4, maybe an esp32-wrover that includes psram solve the problem.

closed 2024-02-25 by github-actions[bot]