schreibfaul1/ESP32-audioI2S #538: Is there a way to skip certificate validation when using https url for file streaming

0x0fe, 2023-05-27:
So, i am streaming mp3 file from an aws storage, in https, however due to the way aws ssl layer is setup it leads to failure in the certificate validation or unreasonnable memory usage when it (very rarely) works. This huge memory need is presumably to store a 16KB certificate, at least i assume so.  I already set my stack to 24K when i normally use only about 3K, and i have 2M PSRAM enable, which is used by esp32-audioi2s, but still the error prevents streaming. On other storages i dont have any error and the HTTPS sets itself up quickly and without problem, i suspect these other storages uses a more reasonnably sized certificate.

So i would like to bypass the certificate validation. I know this is possible because i also download the same files in the background to store them on sd_mmc, and for this i used HTTPClient, still providing and https url, instead of using HTTPSecureClient, which fails (even when i provide the valid certificate).

I know there is also an option like . setInsecure() for HTTPSecureClient which presumably allows to bypass the certificate validation while keeping the encryption.
So my question : is there a way to either force esp32-audioi2s to use HTTPClient instead of HTTPSecureClient, or  to enable the .setIsecureOption on HTTPSecureClient?

Thanks.

schreibfaul1, 2023-05-27:
In menuconfig there is the item: -> component config -> mbedTLS -> memory allocation strategy
there change from 'Internal Memory' to External SPIRAM'
Maybe that will help. Here is an example of an Arduino framework using EspIdf https://github.com/schreibfaul1/ESP32_Arduino_ESPIDF

0x0fe, 2023-05-27:
oh,  isee, i will change that and recompile the core (i dont use platformio)
That is the error i got, despite there is some stack left:

```
22:35:13.239 -> Stack Space: 23848
22:35:13.239 -> Used PSRAM: 59512

```

```
2:35:18.787 -> req 02190530001500
22:35:21.161 -> info        Connect to new host: "https://myfaba-staging-v2.s3.eu-west-3.amazonaws.com/cc_11/l%27elefante_con_le_ghette/8RxQe-1-%20L%27elefante%20con%20le%20ghette.faba?X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIA2PQGQXRMQEQLOHWM%2F20230527%2Feu-west-3%2Fs3%2Faws4_request&X-Amz-Date=20230527T203520Z&X-Amz-SignedHeaders=host&X-Amz-Expires=3600&X-Amz-Signature=27536771d891a16ac7f4345043329f5c5ee11a42b50b32193ae1a6eecf4ebb65"
22:35:21.206 -> info        PSRAM found, inputBufferSize: 283615 bytes
22:35:21.206 -> info        buffers freed, free Heap: 130328 bytes
22:35:21.391 -> [ 12431][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-16256) PK - Memory allocation failed
22:35:21.391 -> [ 12432][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -16256
22:35:21.391 -> info        Request https://myfaba-staging-v2.s3.eu-west-3.amazonaws.com/cc_11/l%27elefante_con_le_ghette/8RxQe-1-%20L%27elefante%20con%20le%20ghette.faba?X-Amz-Content-Sha256=UNSIGNED-PAYLOAD&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIA2PQGQXRMQEQLOHWM%2F20230527%2Feu-west-3%2Fs3%2Faws4_request&X-Amz-Date=20230527T203520Z&X-Amz-SignedHeaders=host&X-Amz-Expires=3600&X-Amz-Signature=27536771d891a16ac7f4345043329f5c5ee11a42b50b32193ae1a6eecf4ebb65 failed!
22:35:21.437 -> station     
22:35:21.437 -> streamtitle 
22:35:21.437 -> icyurl      
22:35:23.168 -> CP01.mp3 [2060KB]
22:35:47.831 -> Done [24s 85KB/s]
22:35:51.130 -> CP02.mp3 [1234KB]
22:36:03.997 -> Done [12s 102KB/s]
22:36:07.431 -> CP03.mp3 [1381KB]
22:36:25.883 -> Done [18s 76KB/s]
22:36:29.218 -> CP04.mp3 [2561KB]
```

closed 2024-02-24 by github-actions[bot]

drego85, 2024-03-17:
Ciao @0x0fe, I was looking for information about the NFC TAG you used in this test! 😬 Were you able to get an MP3 backup of the stories? Thanks and sorry for the OT.

0x0fe, 2024-03-19:
@drego85 
Hello, mmh, this issue was related to certificate and the problem was solved later on.