schreibfaul1/ESP32-audioI2S #539: Better_WiFi_throughput example crashed with watchdog triggered on task 1

0x0fe, 2023-05-27:
From a newly setup PIO, i just build the [Better_WiFi_throughput](https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput) example and flashed to an ESP32D0WDR2 (V3) and got watdog triggered on task 1.

[image]

schreibfaul1, 2023-05-28:
Jumping back to the Arduino Loop Task takes too long.
For testing you can disable the WDT1 in
menuconfig -> component config -> ESP System Settings

0x0fe, 2023-05-28:
thank you, so i assume it should be disabled during bootup and then reenabled just at the end of the setup.

closed 2023-05-28 by 0x0fe