schreibfaul1/ESP32-audioI2S #540: Does not play sequential mp3 files directory

ajacobs-github, 2023-05-28:
I have setup a directory on the SD card with multiple MP3 files.  I start playing the first MP3 file in Setup, then in audio_eof_mp3 find the next MP3 file in the directory and use connecttoFS to start playing that file.  The first two files play correctly, then I get error "audioHeader reading timeout".

platformio.ini:
`
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
upload_port = COM11
monitor_speed = 115200

lib_deps =
    https://github.com/schreibfaul1/ESP32-audioI2S
`

main.cpp
`

#include <Arduino.h>
#include "Audio.h"
#include "SD.h"
#include "SPI.h"
#include "FS.h"

Audio audio;

// Steup SD Card
#define SD_CS     10
#define SD_MOSI   11
#define SD_CLK    12
#define SD_MISO   13
SPIClass spiSD(HSPI);

// Setup I2S
#define I2S_BCLK 6
#define I2S_LRC  7
#define I2S_DOUT 19

// prototypes
void printDirectory(File dir, int numTabs);
void audio_eof_mp3(const char *info);
String audioDir;
File dirOpen;

void setup() {
  Serial.begin(112500);

  // Setup SD Card
  Serial.println("\nSD Card: Initializing ... ");
  Serial.print("  SD_CLK Pin  "); Serial.println(SD_CLK);
  Serial.print("  SD_MISO Pin  "); Serial.println(SD_MISO);
  Serial.print("  SD_MOSI Pin  "); Serial.println(SD_MOSI);
  Serial.print("  SD_CS   Pin  "); Serial.println(SD_CS);
  
  spiSD.begin(SD_CLK, SD_MISO, SD_MOSI, SD_CS);

  if (!SD.begin(SD_CS, spiSD)) {
    Serial.println("  FAILED");
  }
  else {
    Serial.println("  Success");
    uint8_t cardType = SD.cardType();
    if (cardType == CARD_NONE) {
      Serial.println("  SD Card: Error, no SD card");
      return;
    }
    else {
      Serial.print("SD Card Type: ");
      if (cardType == CARD_MMC) {
        Serial.println("MMC");
      } else if (cardType == CARD_SD) {
        Serial.println("SDSC");
      } else if (cardType == CARD_SDHC) {
        Serial.println("SDHC");
      } else {
        Serial.println("UNKNOWN");
      }
      uint64_t cardSize = SD.cardSize() / (1024 * 1024);
      Serial.printf("SD Card Size: %lluMB\n", cardSize);

      // List Files on SD Card
      audioDir = "/mp3";
      dirOpen = SD.open(audioDir);
      Serial.printf("\nDirectory: %s\n", audioDir);
      printDirectory(dirOpen, 0);
      Serial.println("SD Card: End of List\n");
      dirOpen.close();
    }
  }

  // Audio Setup
  Serial.println("\nAudio I2s: Initializing ... ");
  Serial.print("  I2S_BCLK Pin  "); Serial.println(I2S_BCLK);
  Serial.print("  I2S_LRC Pin  "); Serial.println(I2S_LRC);
  Serial.print("  I2S_DOUT Pin  "); Serial.println(I2S_DOUT);

  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(21); // 0...21

  dirOpen = SD.open(audioDir);
  File audioFile =  dirOpen.openNextFile();
  if (audioFile) {
    String filePath = audioDir + "/" + String(audioFile.name());
    if (audio.connecttoFS(SD, filePath.c_str())) {
      Serial.print("File Opened: "); Serial.println(filePath);
    }
    else {
      Serial.print("ERROR: Failed to open file"); Serial.println(filePath);
    }
  }
  else {
    Serial.printf("ERROR: Directory %s is empty", audioDir);
  }

  Serial.println("\nSetup Done");

}

void loop() {

  audio.loop();

}

void printDirectory(File dir, int numTabs) {
  while (true) {

    File entry =  dir.openNextFile();

    if (! entry) {
      // no more files
      break;
    }
    for (uint8_t i = 0; i < numTabs; i++) {
      Serial.print('\t');
    }
    Serial.print(entry.name());
    if (entry.isDirectory()) {
      Serial.println("/");
      printDirectory(entry, numTabs + 1);
    } else {
      // files have sizes, directories do not
      Serial.print("\t\t\t\t");
      Serial.println(entry.size(), DEC);
    }
    entry.close();

  }
}

void audio_eof_mp3(const char *info){  //end of file
  Serial.printf("End of file: %s\n", info);

  File audioFile =  dirOpen.openNextFile();

  Serial.print("Next File: "); Serial.println(audioFile.name());

  if (audioFile) {
    String filePath = audioDir + "/" + String(audioFile.name());
    if (audio.connecttoFS(SD, filePath.c_str())) {
      Serial.print("File Opened: "); Serial.println(filePath);
    }
    else {
      Serial.print("ERROR: Failed to open file"); Serial.println(filePath);
    }
  }
  else {
    Serial.printf("End of Directory");
  }

}

`

Terminal Output:
Audio I2s: Initializing ...
  I2S_BCLK Pin  6
  I2S_LRC Pin  7
  I2S_DOUT Pin  19
File Opened: /mp3/Bottle Empty.mp3

Setup Done
End of file: Bottle Empty.mp3
Next File: Drop Admin.mp3
File Opened: /mp3/Drop Admin.mp3
End of file: Drop Admin.mp3
Next File: Instruction.mp3
File Opened: /mp3/Instruction.mp3
[  5672][E][Audio.cpp:2810] processLocalFile(): audioHeader reading timeout

schreibfaul1, 2023-05-28:
I believe only two files can be open on an SD card at a time.

````c++
#include "Arduino.h"
#include "Audio.h"
#include "SD_MMC.h"
#include "FS.h"

#define I2S_LRC     26
#define I2S_DOUT    25
#define I2S_BCLK    27

#define SD_MMC_D0   2
#define SD_MMC_CLK  14
#define SD_MMC_CMD  15

void getNextFile(); //proto

Audio audio;

File dir;
const char audioDir[] = "/mp3";

void setup() {
    Serial.begin(115200);
    pinMode(SD_MMC_D0, INPUT_PULLUP);
    SD_MMC.setPins(SD_MMC_CLK,SD_MMC_CMD, SD_MMC_D0);
    if(!SD_MMC.begin( "/sdmmc", true, false, 20000)){
        Serial.println("Card Mount Failed");
        return;
    }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, 0);
    audio.setVolume(17); // 0...21 Will need to add a volume setting in the app
    dir = SD_MMC.open(audioDir);
    getNextFile();
}

void loop(){
    audio.loop();
}

void getNextFile(){
    File file = dir.openNextFile();
    if(!file) return;
    Serial.printf("next file: %s\n", file.path());
    String audioFile = file.path();
    file.close();
    audio.connecttoFS(SD_MMC, audioFile.c_str());
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
    getNextFile();
}
```

ajacobs-github, 2023-05-29:
Is there a way to read the next file name without opening the file on the SD card?
-------- Original message --------From: Wolle ***@***.***> Date: 5/28/23  3:50 PM  (GMT-07:00) To: schreibfaul1/ESP32-audioI2S ***@***.***> Cc: ajacobs-github ***@***.***>, Author ***@***.***> Subject: Re: [schreibfaul1/ESP32-audioI2S] Does not play sequential mp3 files from directory (Issue #540) 
I believe only two files can be open on an SD card at a time.
#include "Arduino.h"
#include "Audio.h"
#include "SD_MMC.h"
#include "FS.h"

#define I2S_LRC     26
#define I2S_DOUT    25
#define I2S_BCLK    27

#define SD_MMC_D0   2
#define SD_MMC_CLK  14
#define SD_MMC_CMD  15

void getNextFile(); //proto

Audio audio;

File dir;
const char audioDir[] = "/mp3";

void setup() {
    Serial.begin(115200);
    pinMode(SD_MMC_D0, INPUT_PULLUP);
    SD_MMC.setPins(SD_MMC_CLK,SD_MMC_CMD, SD_MMC_D0);
    if(!SD_MMC.begin( "/sdmmc", true, false, 20000)){
        Serial.println("Card Mount Failed");
        return;
    }
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT, -1, 0);
    audio.setVolume(17); // 0...21 Will need to add a volume setting in the app
    dir = SD_MMC.open(audioDir);
    getNextFile();
}

void loop(){
    audio.loop();
}

void getNextFile(){
    File file = dir.openNextFile();
    if(!file) return;
    Serial.printf("next "pl-cce">\n", file.path());
    String audioFile = file.path();
    file.close();
    audio.connecttoFS(SD_MMC, audioFile.c_str());
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
    getNextFile();
}
```

—Reply to this email directly, view it on GitHub, or unsubscribe.You are receiving this because you authored the thread.Message ID: ***@***.***>

schreibfaul1, 2023-05-29:
You can use a vector for this.
https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/plays%20all%20files%20in%20a%20directory/plays%20all%20files%20in%20a%20directory.ino

ajacobs-github, 2023-05-30:
Thank you @schreibfaul1.  The issue seemed to be too many files open.  When I closed the file after openNextFile and before sending to connecttoFS, it works well.

`void playNextAudio() {
  File audioFile =  audioDir.openNextFile();
  if (!audioFile) {
    audioDir.rewindDirectory();    
    audioFile =  audioDir.openNextFile();
    if(!audioFile) {
      Serial.print("\nDirectory empty");
      return;
    } 
  }
  String filePath = audioFile.path();
  audioFile.close();
  if (audio.connecttoFS(SD, filePath.c_str())) {
    timeAudioStart = millis();
    Serial.print("\nPlaying Audio File: "); Serial.println(filePath);
  }
  else {
    Serial.print("\nERROR: Failed to open file"); Serial.println(filePath);
  }

}`

closed 2024-02-11 by github-actions[bot]