schreibfaul1/ESP32-audioI2S #541: Some issues on platformio (which do no occur on Arduino)

0x0fe, 2023-05-31:
I had to move a firmware from arduino to PIO in order to have a decently easy way to modify the IDF configuration, in particular for allowing memory allocation on PSRAM (SPIRAM) for mbedTLS, wifi and BT stack, and also to improve the wifi throughput.
However to my surprise, what was perfectly working on arduino (2.0.8) proved to not work at all on PIO, i made a test firmware to demonstate the problem.

The test firmware is quite straightorward : 
One thread start an audio streaming/decoding to I2S of an mp3 file (HTTPS) 
Another thread starts a download of another (or the same) file, and save it to SD card which is in SDIO 1-bit mode.
This continues in loop, each time a downlaod is finished it starts a new one, each time a stream is finished it starts a new one, both are rolling a set of 3 urls.

The board is fitted with ESP32D0WDR2 which is identical to the "legacy" ESP32D except it has 2MB of internal PSRAM.
It has an audio codec rather than a simple DAC (for various reasons) however i believe the problem can be reproduced with any DAC, it has nothing to do with the specific codec i use. In the code below i commented the calls specific to my codec, so that it can be tested with any DAC.

The issues : 

The first issue is : when i declare the Audio audio object normally (on top of the main.cpp) the esp will just crash at early boot, complaning about the sample rate, incorrect pins and problem with MCLK. Strangely enough it will mke the sd_mmc to systematically crash as well (probably related to DMA). I am not sure what causes this because in Arduino the exact same firmware and library are used, and there is no problem at all with the initialisation of the library. I believe this has to do with how PIO has implemented the early calls, and also hwo it manages the setup() function, which is clearly differnt from how it is managed in arduino (hence the need to disable watchdog on core 1, which is not the case for arduino). I suspect maybe these object are initialized before the I2S driver or the PSRAM is ready for example.
Anyway to go ahread i tried to hardcode the pins number, samplerate and set a fixed MCLK configuration directly in Audio::Audio, which seems to work since i got no error reported after this change.

```cpp
    clientsecure.setInsecure();  // if that can't be resolved update to ESP32 Arduino version 1.0.5-rc05 or higher
    m_f_channelEnabled = channelEnabled;
    m_f_internalDAC = internalDAC;
    //i2s configuration
    m_i2s_num = i2sPort; // i2s port number
    m_i2s_config.sample_rate          = 48000;//16000;
    m_i2s_config.bits_per_sample      = I2S_BITS_PER_SAMPLE_16BIT;
    m_i2s_config.channel_format       = I2S_CHANNEL_FMT_RIGHT_LEFT;
    m_i2s_config.intr_alloc_flags     = ESP_INTR_FLAG_LEVEL1; // high interrupt priority
    m_i2s_config.dma_buf_count        = 8;      // max buffers
    m_i2s_config.dma_buf_len          = 1024;   // max value
    m_i2s_config.use_apll             = 0 ;//APLL_DISABLE; // must be disabled in V2.0.1-RC1
    m_i2s_config.tx_desc_auto_clear   = true;   // new in V1.0.1
    m_i2s_config.fixed_mclk           = 48000*256; //44100*256;//I2S_PIN_NO_CHANGE;

    m_pin_config.mck_io_num = 3;
    m_pin_config.bck_io_num = 18;
    m_pin_config.data_out_num = 19;
    m_pin_config.ws_io_num = 5;
```

Then comes the other issue, when i modify this, i got no error at init, and everything seems to work correctly (stream, download) however, there is no audio, at all.

```
.........
link up
IP: 192.168.1.25
init_sd
Type: SDHC
Total: 31968MB
Used: 1807MB
init_audio
[  1519][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=4 scl=22 freq=100000
I2C ready
SDA:4 SCL:22 Freq:100KHz
NAU88C22 [0x1A]
HP vol:90
Spk vol:89
start stream
info PSRAM found, inputBufferSize: 283615 bytes
info buffers freed, free Heap: 134335 bytes
info Connect to new host: "https://files.freemusicarchive.org/storage-freemusicarchive-org/tracks/EF6VfQVDlEVfkAhWGvwA3YFKFALkTOg8HdNogCHY.mp3"
info Connect to "files.freemusicarchive.org" on port 443, extension "/storage-freemusicarchive-org/tracks/EF6VfQVDlEVfkAhWGvwA3YFKFALkTOg8HdNogCHY.mp3"
start download
info SSL has been established in 1195 ms, free Heap: 130543 bytes
info HTTP/1.1 200 OK
info date: Wed, 31 May 2023 15:38:51 GMT
info content-type: audio/mpeg, format is mp3
info MP3Decoder has been initialized, free Heap: 107615 bytes
info content-length: 8413648
info last-modified: Wed, 05 Apr 2023 02:43:43 GMT
info etag: "642ce05f-8061d0"
info access-control-allow-origin: *
[... 410 lines cut ...]
      audio.connecttohost(urls[playIndex]); 
      if(++playIndex>=URL_CNT)playIndex=0;
    }
    audio.loop();  
  }
}
void setup(void){

  disableCore1WDT();
  disableCore0WDT();
  Serial.begin(115200,SERIAL_8N1,-1,1); 
  //startMemMon();
  //delay(5000);//PIO takes like 5000ms to reopen the uart for monitoring after upload :(
  init_wifi(); 
  init_sd();
  xTaskCreate(audio_task, "stream", 13500, NULL, 15, NULL);  
  play=1;
}
void loop(void){
  
  if(down){down=0; xTaskCreatePinnedToCore(download_task,"downloader",13000,NULL,2,NULL,1);}
  //log_handler();
}
#endif 
```

0x0fe, 2023-05-31:
i solved the mystery of audio playback issue, apparently it was necessary to call the setpinout and  functions for set_mclk_pin_select even though they were already hardcoded in the library.
Now the audio is back.

```
  audio.setPinout(BCK, WS, DAT);
  audio.i2s_mclk_pin_select(MCK);
```

pwilkowski, 2023-06-01:
I have similar problem and his solution does not work for me, however i have found some values that does not produce sample rate too high error. MCU still restarts

imo, this library should expose `m_i2s_config` in some way

I am currently trying
```cpp
    m_i2s_num = i2sPort; // i2s port number
    m_i2s_config.sample_rate          = 44100; //48000;//16000;
    m_i2s_config.bits_per_sample      = I2S_BITS_PER_SAMPLE_16BIT;
    m_i2s_config.channel_format       = I2S_CHANNEL_FMT_RIGHT_LEFT;
    m_i2s_config.intr_alloc_flags     = ESP_INTR_FLAG_LEVEL1; // high interrupt priority
    m_i2s_config.dma_buf_count        = 8;      // max buffers
    m_i2s_config.dma_buf_len          = 1024;   // max value
    m_i2s_config.use_apll             = APLL_DISABLE; // must be disabled in V2.0.1-RC1
    m_i2s_config.tx_desc_auto_clear   = true;   // new in V1.0.1
    m_i2s_config.mclk_multiple   	  = I2S_MCLK_MULTIPLE_DEFAULT;   // new in V1.0.1
    m_i2s_config.fixed_mclk           = 0;//I2S_PIN_NO_CHANGE; //48000*256;//
```

0x0fe, 2023-06-10:
@schreibfaul1 
yes that is what i ended up doing, having a separate tasks and communicating with it via message queue.
I noticed that the MP3 EOF callabck has to call direct function though, otherwise it locks up, but it makes sense since this callback is actually called from the thread itself.

@pwilkowski 
You can do what i mention, it is stable. Just be careful to call functions directly when the caller is the audio thread, in all other cases use mesasge queues. 

Regarding the issues with Setup in PIO, i assume the setup function is not called as early as in arduino implementation and for this reason the WDT can easily get triggered. My solution for now is to initilize most things in their own thread, with basic signaling to avoid run conditions.
There is still problems with PIO but msot are solved when using a spearate thread and messagequeues.

schreibfaul1, 2023-06-10:
I'm not sure if I understood the question correctly. For example, if EOF performs an action on a device that is already occupied by another task, collisions will occur and strange errors will occur.
To avoid this, a task must wait until the device is free.  Semaphores are usually used for this.

````c++
SemaphoreHandle_t  mutex_dev;

void setup(){
...
mutex_dev = xSemaphoreCreateMutex();
}

void do_something(){
    xSemaphoreTake(mutex_dev, portMAX_DELAY);
    // .....
    xSemaphoreGive(mutex_dev);
}

void audio_eof_mp3(const char *info){
     do_something();
}
````

closed 2024-02-10 by github-actions[bot]