schreibfaul1/ESP32-audioI2S #543: Does this library work with ES8311? Specifically ESPS3-BOX?

djschwab, 2023-06-01:
Does this library work with ES8311? Specifically ESPS3-BOX? I tried the following, but no sound was produced:
// Digital I/O used
#define SD_CS         29 // 5
#define SPI_MOSI      32 // 23
#define SPI_MISO      31 // 19
#define SPI_SCK       17 // 18
#define I2S_DOUT      15 // 25
#define I2S_BCLK      2 // 27
#define I2S_LRC       47 // 26
Audio audio;

schreibfaul1, 2023-06-01:
An external library for control via I2C is required for the ES8311. Maybe there is one that can be used, or make your own according to the datasheet.

djschwab, 2023-06-01:
Thank you for your reply.

schreibfaul1, 2023-06-01:
maybe you can adapt this:
https://github.com/pschatzmann/arduino-audiokit/tree/main/src/audio_driver/es8311

closed 2024-02-09 by github-actions[bot]