schreibfaul1/ESP32-audioI2S #545: Audio is very choppy and disconnects often

WifiRouterYT, 2023-06-04:
Hey everyone! Trying to make an internet radio out of some spare parts, and this is the only library I've had success with for streaming audio from the web, however the audio is very inconsistent. Sometimes its smooth as butter and sounds great, other times it gets very choppy and starts to disconnect/reconnect a bunch. Is there a fix to this? I know there's a page in this repo for improving WiFi and stuff but I'm using the Arduino IDE which does not have menuconfig, so if there's another way, that'd be great.

WifiRouterYT, 2023-06-04:
Serial output:
[image]

schreibfaul1, 2023-06-05:
The precompiled Arduino only allows limited WiFi throughput. It works better with changed TCP values ​​in the menuconfig, see https://github.com/schreibfaul1/ESP32-audioI2S/tree/master/examples/Better_WiFi_throughput

WifiRouterYT, 2023-06-05:
yes i saw that - im just confused on how to access menuconfig through the Arduino IDE.

schreibfaul1, 2023-06-05:
I don't think it works with the Arduino IDE. This is a demo for PlatformIO.

WifiRouterYT, 2023-06-05:
ah. unfortunate

tv4you2016, 2023-06-09:
@WifiRouterYT 

hello you have to make your own build

https://github.com/platformio/platform-espressif32

altJSV, 2023-06-26:
> Hey everyone! Trying to make an internet radio out of some spare parts, and this is the only library I've had success with for streaming audio from the web, however the audio is very inconsistent. [...]

Doesn't editing this file C:\Users\ username \AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.9\tools\sdk\esp32\sdkconfig affect anything?

closed 2024-02-08 by github-actions[bot]