schreibfaul1/ESP32-audioI2S #553: Vorbis decoder not handling continued pages correctly

ma261065, 2023-06-16:
I don't think the Vorbis Decoder is handling pages with the Continuation flag set correctly.

For example - taking this file https://archive.org/download/gd1980-10-29.beyer.stankiewicz.126919.flac1644/gd1980-10-29s1t02.ogg

The first 5 Ogg pages contain Vorbis packets that finish on the boundary of the page. The Continuation flag is not set, and they play correctly.

However, the 6th Ogg page starts 4 bytes before the end of the previous Vorbis packet, cutting the end of the packet off, and as such has the Continuation flag set.

The last 4 bytes of the previous Vorbis packet are in the first segment of Ogg page 6. However, the decoder skips this whole Ogg page (all 255 segments), leading to an audible gap in the audio.

A hex dump might make it easier to understand:

`...`
`00020e20  68 34 3e 6a 70 b3 3f 30  6a 20 ce b1 85 4f d6 20  |h4>jp.?0j ...O. |`
`00020e30  5e 23 71 9e be 11 25 de  68 fb c8 5f 48 c1 45 1f  |^#q...%.h.._H.E.|`
`00020e40  59 82 45 4a ce 91 3d 01  f3 57 02 bd 86 fa d2 0b  |Y.EJ..=..W......|`
`00020e50  2b 3d 1b 5e 7a e5 b9 33  21 1b 2f 35 3e f0 4e b3  |+=.^z..3!./5>.N.|` 
Last segment in OGG Page 5. Starts at 0x20e53, Length = 0x103. Should finish at 0x20f55 (Vorbis Packetno 594)
`00020e60  5b 84 f7 2a e5 c8 41 ee  f1 e1 2a b2 01 9d 81 43  |[..*..A...*....C|`
`00020e70  09 78 89 2d 10 58 02 15  92 e0 6d 7a 34 1c 21 7a  |.x.-.X....mz4.!z|`
`00020e80  78 54 0d b7 87 f7 48 c4  88 fa df db 13 ed e3 1b  |xT....H.........|`
`00020e90  17 4e da 5b eb ce 29 89  58 92 1c f4 92 ea a2 9c  |.N.[..).X.......|`
`00020ea0  8d 9c eb eb ef c5 6e 67  f9 1f 06 76 c8 db d6 a2  |......ng...v....|`
`00020eb0  e4 46 30 64 00 20 eb d5  c7 13 dc 5b df b5 09 cc  |.F0d. .....[....|`
`00020ec0  6c 69 47 0e cd ff dc ad  28 98 1d fe f7 99 22 fa  |liG.....(.....".|`
`00020ed0  69 92 d4 c2 dd c1 79 5b  73 8a eb e6 7a 91 af bc  |i.....y[s...z...|`
`00020ee0  d0 f9 de 02 d5 8f 66 9c  fa 69 02 49 10 24 d3 07  |......f..i.I.$..|`
`00020ef0  38 ed 0a c8 05 05 39 fe  35 20 bb 7e 77 a5 e5 06  |8.....9.5 .~w...|`
`00020f00  e0 c6 1f f6 9a 01 f0 4d  a2 c8 bc f9 a4 92 d6 b7  |.......M........|`
`00020f10  04 00 b8 9a 03 58 ef 8f  5c 5b 2d ba fd 34 06 d9  |.....X..\[-..4..|`
`00020f20  e0 6d 95 70 f9 7c 50 d2  48 83 d5 04 06 c1 ad 34  |.m.p.|P.H......4|`
`00020f30  74 75 a7 e6 c6 2e 3a 6d  ca 77 4b fd f5 d3 ae 32  |tu....:m.wK....2|`
`00020f40  7c ad e4 77 35 1a 10 c9  5f 11 91 dd 8c 03 2a d7  ||..w5..._.....*.|` 
Segment finishes at 0x20f51 with 4 bytes to go
`00020f50  53 d0 4f 67 67 53 00 01  40 e0 08 00 00 00 00 00  |S.OggS..@.......|`
Ogg Page 6 starts at 0x20f52:  4f 67 67 53 = OggS, 00 = version, 01 = header type: Continuation, 40 e0 08 00 00 00 00 00 = Granule Pos
`00020f60  90 56 ae 30 06 00 00 00  22 0c 5c 5b ff 04 ff 02  |.V.0....".\[....|`
90 56 ae 30 = Bitstream Serial, 06 00 00 00 = Page Sequence, 22 0c 5c 5b = CRC, ff = num segments (255)
`00020f70  ff 0a ff 15 ff 02 ff 03  ff 0f ff 0f ff 13 ff 0d  |................|`
Segment sizes 04 ff 02 ff 0a ff 15 ff 02 ff 03 ... 07 ff
`00020f80  ff 05 ff 17 ff 19 ff 08  ff 22 30 32 31 36 33 3f  |........."02163?|`
`00020f90  43 48 3c 3a 35 41 46 ff  1b ff 17 ff 0d ff 14 ff  |CH<:5AF.........|`
`00020fa0  15 ff 19 ff 1d ff 12 ff  1d ff 13 ff 0b ff 0e ff  |................|`
`00020fb0  0a ff 1a 41 4b ff 1b 32  34 31 30 2f 3d 41 44 33  |...AK..2410/=AD3|`
`00020fc0  31 36 37 32 42 40 ff 15  ff 05 ff 05 ff 15 ff 1d  |1672B@..........|`
`00020fd0  ff 1a ff 0d ff 0f ff 17  ff 15 ff 1c ff 06 ff 0e  |................|`
`00020fe0  ff 0e ff 10 ff 17 34 31  34 31 38 33 40 4d 47 34  |......414183@MG4|`
`00020ff0  30 33 35 48 46 50 ff 11  ff 16 ff 0f ff 15 ff 18  |035HFP..........|`
`00021000  ff 16 ff 14 ff 0c ff 1a  ff 0d ff 00 ff 16 ff 17  |................|`
`00021010  35 33 34 36 43 50 ff 25  ff 0e ff 0d ff 0b ff 11  |5346CP.%........|`
`00021020  ff 15 ff 0d ff 09 ff 0e  ff 05 ff 14 ff 1e ff 19  |................|`
`00021030  ff 12 ff 0c ff 0d ff 0d  ff 01 ff 09 ff 11 ff 11  |................|`
`00021040  ff 11 ff 13 ff 10 ff 15  ff 06 ff 07 ff 12 ff 18  |................|`
`00021050  ff 06 ff 0a ff 05 f5 ff  00 ff 0f 37 33 33 32 41  |...........7332A|`
`00021060  4e ff 12 ff 09 ff 12 ff  0d ff 07 ff 60 0a 8a 03  |N...............|`
Segment 1. Starts 0x2106c, Size = 0x04 (Data 60 0a 8a 03). This is the last 4 bytes of the previous Vorbis packet.
`00021070  7e 8a 55 c8 35 42 2f ad  d7 59 4d 32 b1 5c 05 ed  |~.U.5B/..YM2.\..|`
Segment 2. Starts 0x21070, Size = 0x101 (Vorbis Packetno 595)
`00021080  f1 f8 71 60 c1 5f 75 a0  1d 58 42 35 10 26 22 11  |..q.._u..XB5.&".|`
`00021090  5b 19 8a fc 54 7c 0d b7  4d 44 b4 3e 5f a9 ac 21  |[...T|..MD.>_..!|`
`000210a0  3f c8 d8 3c 23 63 bd 8f  1c a5 9b 57 9b 63 eb 5f  |?..<#c.....W.c._|`
`000210b0  fb 8b c5 fb d9 7f bf bb  bd f0 f3 b1 8f af dd f6  |................|`
`000210c0  de de 8f 4e 3a eb 3c af  fe f9 63 9b f3 c5 8c dd  |...N:.<...c.....|`
`000210d0  74 c9 c6 bb b9 5f 43 ec  10 88 2f f1 e6 b5 80 5c  |t...._C.../....\|`
`000210e0  41 96 2d ef 3c 13 08 b5  e1 ac 09 a1 72 d4 a5 2b  |A.-.<.......r..+|`
`000210f0  3b f8 d6 90 8b 50 e0 f1  d7 7f bb 0c ea ec b1 dd  |;....P..........|`
`00021100  a7 e2 70 2a d2 ae 17 03  c8 4e 2c 72 d7 17 b8 48  |..p*.....N,r...H|`
`00021110  b6 0b f4 7f 20 d2 1f f6  06 ad 20 5a 14 c4 9f ff  |.... ..... Z....|`
`00021120  f6 3e 02 ec c6 ac 4f 6e  7e 23 02 c0 0f f8 ed ff  |.>....On~#......|`
`00021130  08 00 80 c7 c0 04 01 03  a4 39 1b 40 01 a4 c9 47  |.........9.@...G|`
`00021140  10 80 39 02 a0 0d 02 75  0f 48 e4 67 c1 57 fa 05  |..9....u.H.g.W..|`
`00021150  e9 b0 4a 15 91 44 c7 e3  1c 7d 97 6d 13 8a 67 58  |..J..D...}.m..gX|`
`00021160  16 5c ab 53 ab c1 aa 75  03 79 85 33 04 0b 43 ca  |.\.S...u.y.3..C.|`
`00021170  01 be 7a 45 e4 99 a0 99  0f 4b 5c 0c f9 e6 58 2d  |..zE.....K\...X-|`
Segment 3. Starts 0x22171, Size = 0x109. (packetno 596)
`00021180  c1 8e 10 17 d4 ca e4 c5  03 da 69 13 b7 57 ac ac  |..........i..W..|`
`00021190  97 b8 78 56 50 02 bc 6a  60 09 2e 04 db 60 21 d9  |..xVP..j......!.|`
`000211a0  00 f3 f5 82 1b 03 a3 1d  11 29 fb 42 33 6e 18 dc  |.........).B3n..|`
`...`

ma261065, 2023-06-16:
For comparison, this is the output of the `oggz dump` command, showing the relevant packets:

`00:00:10.746: serialno 0816731792, calc. gpos 473920, packetno 594: 259 bytes`
`0000: 5e7a e5b9 3321 1b2f 353e f04e b35b 84f7  ^z..3!./5>.N.[..`
`0010: 2ae5 c841 eef1 e12a b201 9d81 4309 7889  *..A...*....C x.`
`0020: 2d10 5802 1592 e06d 7a34 1c21 7a78 540d  -.X....mz4.!zxT`
`0030: b787 f748 c488 fadf db13 ede3 1b17 4eda  ...H..........N.`
`0040: 5beb ce29 8958 921c f492 eaa2 9c8d 9ceb  [..).X..........`
`0050: ebef c56e 67f9 1f06 76c8 dbd6 a2e4 4630  ...ng...v.....F0`
`0060: 6400 20eb d5c7 13dc 5bdf b509 cc6c 6947  d. .....[.. .liG`
`0070: 0ecd ffdc ad28 981d fef7 9922 fa69 92d4  .....(.....".i..`
`0080: c2dd c179 5b73 8aeb e67a 91af bcd0 f9de  ...y[s...z......`
`0090: 02d5 8f66 9cfa 6902 4910 24d3 0738 ed0a  ...f..i.I.$..8.`
`00a0: c805 0539 fe35 20bb 7e77 a5e5 06e0 c61f  ...9.5 .~w......`
`00b0: f69a 01f0 4da2 c8bc f9a4 92d6 b704 00b8  ....M...........`
`00c0: 9a03 58ef 8f5c 5b2d bafd 3406 d9e0 6d95  ..X..\[-..4...m.`
`00d0: 70f9 7c50 d248 83d5 0406 c1ad 3474 75a7  p.|P.H......4tu.`
`00e0: e6c6 2e3a 6dca 774b fdf5 d3ae 327c ade4  ...:m.wK....2|..`
`00f0: 7735 1a10 c95f 1191 dd8c 032a d753 d060  w5..._.....*.S..`
`0100: 0a8a 03                                  ...`

`00:00:10.769: serialno 0816731792, calc. gpos 474944, packetno 595: 257 bytes`
`0000: 7e8a 55c8 3542 2fad d759 4d32 b15c 05ed  ~.U.5B/..YM2.\..`
`0010: f1f8 7160 c15f 75a0 1d58 4235 1026 2211  ..q.._u..XB5.&".`
`0020: 5b19 8afc 547c 0db7 4d44 b43e 5fa9 ac21  [...T| .MD.>_..!`
`0030: 3fc8 d83c 2363 bd8f 1ca5 9b57 9b63 eb5f  ?..<#c.....W.c._`
`0040: fb8b c5fb d97f bfbb bdf0 f3b1 8faf ddf6  ................`
`0050: dede 8f4e 3aeb 3caf fef9 639b f3c5 8cdd  ...N:.<...c.....`
`0060: 74c9 c6bb b95f 43ec 1088 2ff1 e6b5 805c  t...._C.../....\`
`0070: 4196 2def 3c13 08b5 e1ac 09a1 72d4 a52b  A.-.<..... .r..+`
`0080: 3bf8 d690 8b50 e0f1 d77f bb0c eaec b1dd  ;....P..... ....`
`0090: a7e2 702a d2ae 1703 c84e 2c72 d717 b848  ..p*.....N,r...H`
`00a0: b60b f47f 20d2 1ff6 06ad 205a 14c4 9fff  . .. ..... Z....`
`00b0: f63e 02ec c6ac 4f6e 7e23 02c0 0ff8 edff  .>....On~#......`
`00c0: 0800 80c7 c004 0103 a439 1b40 01a4 c947  .........9.@...G`
`00d0: 1080 3902 a00d 0275 0f48 e467 c157 fa05  ..9.. .u.H.g.W..`
`00e0: e9b0 4a15 9144 c7e3 1c7d 976d 138a 6758  ..J..D...}.m..gX`
`00f0: 165c ab53 abc1 aa75 0379 8533 040b 43ca  .\.S...u.y.3. C.`
`0100: 01                                       .`

`00:00:10.792: serialno 0816731792, calc. gpos 475968, packetno 596: 265 bytes`
`0000: be7a 45e4 99a0 990f 4b5c 0cf9 e658 2dc1  .zE.....K\ ..X-.`
`0010: 8e10 17d4 cae4 c503 da69 13b7 57ac ac97  .........i..W...`
`0020: b878 5650 02bc 6a60 092e 04db 6021 d900  .xVP..j. ....!..`
`0030: f3f5 821b 03a3 1d11 29fb 4233 6e18 dc25  ........).B3n..%`
`0040: bfb5 be6e cf1e 6706 cddf d1ed de6d 33f5  ...n..g......m3.`
`0050: 5a9f b7a1 d7b6 bbe5 5c9a c7b0 c296 ec64  Z.......\......d`
`0060: 8760 00c0 f0e5 2842 35c1 4c9a 66fe 68fa  ......(B5.L.f.h.`
`0070: c997 7cb9 b083 2d78 cbf3 53a0 1063 d3c8  ..|...-x..S..c..`
`0080: b402 4a64 7e59 738c 6894 1eb3 0204 f372  ..Jd~Ys.h......r`
`0090: d903 c4bd c536 0f51 e49b 6c80 6342 3e5a  .....6.Q..l.cB>Z`
`00a0: 15aa cc14 8b09 01f8 b228 3e48 0b22 fe2d  ..... ...(>H ".-`
`00b0: 090d 4f4a e243 03b4 fde9 ff44 2388 832e    OJ.C.....D#...`
`00c0: 5520 48a9 f049 0b04 18fb 1f01 0580 f806  U H..I .........`
`00d0: 40d0 0cde 0242 03e0 ee77 4a01 6c00 3405  @. ..B...wJ.l.4.`
`00e0: 7d1f 0121 54d2 5508 d202 2c6d 1059 d3a6  }..!T.U...,m.Y..`
`00f0: 4456 83d6 5999 c601 4e87 3d10 46a5 9701  DV..Y...N.=.F...`
`0100: 550b 50c7 8882 976d 00                   U P....m.`

schreibfaul1, 2023-06-17:
Thanks for the good work. Yes, there is a problem when the 'Continuation flag' is set. I'll take a closer look at that soon.

schreibfaul1, 2023-06-18:
Thanks to your good analysis, I was able to narrow down the problem easily. I hadn't looked at the 'continued page' topic at all before. Many web streams had no errors. I have now added the necessary changes, the example file https://archive.org/download/gd1980-10-29.beyer.stankiewicz.126919.flac1644/gd1980-10-29s1t02.ogg works now.

ma261065, 2023-06-18:
Thank you so much. It does indeed seems to have fixed the error in that test file. I'll try a few more files to be sure, but so far it looks good.

ma261065, 2023-07-01:
A couple of thoughts on this continued pages issue:

The section "Packets spanning pages" at https://xiph.org/vorbis/doc/framing.html mentions a few things of interest:

1) "Note that a packet can span an arbitrary number of pages" Does your solution account for this? I don't know how common it is in the real world, so may be of academic interest only.

2) "... a header flag of 0x01 is set to indicate a continued packet. Although mandatory, it is not actually algorithmically necessary; one could inspect the preceding segment table to determine if the packet is new or continued." - I note that you buffer each packet "just in case" it is continued, and that you allocate an extra buffer just for this purpose, which may not be optimal in memory-constrained environments. Would it be possible to use the existing packet buffer if you algorithmically  determined that a packet was going to be continued, rather than only finding out once you are in a new page?

3) I have found some other files that will not play, but for a different reason. I will create a new issue for these, and you may (or may not) choose to fix it there.

Once again, thank you for all of your hard work in giving us these great audio libraries!!

ma261065, 2023-08-28:
Closing as completed - noting the comment above for some further thoughts that you may want to consider...

closed 2023-08-28 by ma261065