schreibfaul1/ESP32-audioI2S #556: How to install?

d3mac123, 2023-06-29:
Any ideas why I getting this error (sorry for such a dumb question)?

`/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x30): undefined reference to `MP3Decoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x38): undefined reference to `AACDecoder_IsInit()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x3c): undefined reference to `AACDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x40): undefined reference to `FLACDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x44): undefined reference to `OPUSDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x48): undefined reference to `VORBISDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x28): undefined reference to `AACGetFormat()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x2c): undefined reference to `AACGetID()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x30): undefined reference to `AACGetProfile()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x18): undefined reference to `MP3FindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x1c): undefined reference to `AACFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x20): undefined reference to `AACSetRawBlockParams(int, int, int, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x24): undefined reference to `FLACSetRawBlockParams(unsigned char, unsigned int, unsigned char, unsigned int, unsigned int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x28): undefined reference to `FLACFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x2c): undefined reference to `OPUSFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x30): undefined reference to `VORBISFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x1c): undefined reference to `MP3GetBitrate()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x20): undefined reference to `AACGetBitrate()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D412814F955C122B5118C2CA6/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x24): undefined reference to `FLACGetBitRate()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/5D8BF62D4128 [... cut ...] ISDecoder_FreeBuffers()'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "Audio.h"
  Used: /Users/alexsouza/Documents/Arduino/libraries/ESP32-audioI2S
  Not used: /Users/alexsouza/Library/Arduino15/libraries/ESP32-audioI2S-master
Multiple libraries were found for "SD.h"
  Used: /Users/alexsouza/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/libraries/SD
  Not used: /Users/alexsouza/Library/Arduino15/libraries/SD
exit status 1

Compilation error: exit status 1`

schreibfaul1, 2023-06-30:
Sorry if I don't look at this.

Multiple libraries were found for 'Audio.h' ??? 
How can this be, if you copy the link from the github repo and clone the repository in PlatfornIO, there is only one SD.h, and SD.h is not used at all.

d3mac123, 2023-06-30:
I am not using PlatformIO. This is from Arduino IDE

schreibfaul1, 2023-06-30:
It's too much work to copy the files and folders to the Arduino environment. A lot can go wrong with this. PlatformIO knows the correct Arduino and IDF version. Please use PlatformIO with it it's easy. PlatfornIO does most of it for you, you don't need to copy anything into the project and the configuration is done automatically except for the parameters that you set yourself.

closed 2024-02-07 by github-actions[bot]