schreibfaul1/ESP32-audioI2S #558: Vorbis decoder not handling files with non-contiguous pages

ma261065, 2023-07-01:
As an example - this file https://archive.org/download/gd1991-08-13.sbd.miller.109241.flac16/gd91-08-13d1t02.ogg

When looking at the header, it is very strange - page 0 should finish at 0x39, but instead finishes at 0x3ff. I don't know if the encoder (which seems to be Lavc 52.79.1 - https://github.com/libav/libav) is trying to put pages on 1k boundaries or something? And I'm not even sure that it meets the spec, but nonetheless these files are out there.

I see that Google's Exo Player: https://github.com/google/ExoPlayer/issues/7230#issuecomment-634048746 ran across the same issue, and chose to fix it.

The good news is that the fix appears trivial:

In VORBISparseOGG

Line 563: Change `int idx = VORBIS_specialIndexOf(inbuf, "OggS", 6);` to `int idx = VORBIS_specialIndexOf(inbuf, "OggS", 1024);`
Line 564: Replace `if(idx != 0) return ERR_VORBIS_DECODER_ASYNC;` with `inbuf += idx;`
Line 622: Change the `*bytesLeft -= headerSize;` to `*bytesLeft -= headerSize + idx; `

I'll leave it up to you if you want to "fix" the decoder to handle these non-standard files.

schreibfaul1, 2023-07-02:
Hello, what you found is very peculiar. I can't find any explanation in the ogg specification. The whole thing looks like this:
[image]
There are two blocks, the first 0x1D and the second 0x3a8 bytes in size. Both with unknown content. The easiest is to skip the two blocks. I'm not sure if you can do anything with the content.

schreibfaul1, 2023-07-02:
I implemented it the way you suggested. Presumably the 1K limit for 'non-contiguous pages' must be observed. It's the same with the 2nd OGG page. It is strange that the fill bits are not 0xFF or 0x00.
[image]

Thanks for the good analysis and preliminary work!

ma261065, 2023-08-28:
I have seen no issues with these files since this fix was implemented. Thank you for your hard work!!!!

closed 2023-08-28 by ma261065