schreibfaul1/ESP32-audioI2S #562: AGC or automatic gain control

schuh8, 2023-07-06:
Like another poster, I must first thank you for this wonderful library. With it I was able cram a complete internet radio into a package just 75x55x20 mm. Vielen Dank.
I use this every night to promote sleep. Only problem I have is that there is no standard for volume among internet radio stations and on changing stations one might be too low to hear where the next station may hurt your eardrums.
In the days of over the air radio there was a circuit called AGC or automatic gain control which did a good job of holding the volume of different stations relatively constant. Is it possible to incorporate coding which would integrate the volume of the signal over a few seconds and then adjust the volume of the radio  accordingly? 
There are many internet radio projects on the web but I cannot find one that has any sort of AGC coding. I realize that AGC coding may not properly belong within your library, but perhaps you or a fellow enthusiast might point me in the right direction. 
Again, thank you.

schreibfaul1, 2023-07-06:
That's not easy. I think you will have to develop something yourself.
There is a function called like this: `uint16_t level = audio.getVUlevel()`
This is a function to control a VU meter in my radio. In principle, the peak values ​​are recorded from the samples and an average value is formed from them. If you call the function 10 times per second you can easily observe the modulation, It should be noted that the 16-bit value contains the right and left channel. To separate: uint8_t left = (level & 0xFF00) >> 8; uint8_t right = level & 0x00FF; 
Hope this helps you, you could further average the values ​​and then adjust the volume accordingly. By the way, more than 22 volume values ​​are possible: with setVolumeSteps() you can also set 150 values (or 250) ​​in order to be able to fine-tune the volume.

schuh8, 2023-07-06:
Thanks for the tip. I am going to work on it! If I have any success I'll report back!

closed 2024-02-06 by github-actions[bot]