schreibfaul1/ESP32-audioI2S #563: Distorted signal, sound distortion

bhpjean, 2023-07-09:
Why do I hear large distortions on loud sounds in the new library? what has been broken? using the older version is ok. There is an effect like it would be too loud.
I am using PCM5102A

When I compile the program using the older i2s Audio library, the sound quality is very good.

schreibfaul1, 2023-07-11:
A sinus tone 1KHz, 100% modulation at maximum volume:
Audacity
[image]

Oscilloscope at the output of the PCM5102:
[image]
[Testton_1.0.mp3.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/12013722/Testton_1.0.mp3.zip)

bhpjean, 2023-07-11:
I did the signal measurements, for me the results came out like this.
The first oscillogram is for the old library from Dec 31.2021
[image]

the second oscillogram is for the latest library available on github
[image]

I didn't change anything in my internet radio program, I only replaced the ESP32-audioI2S library and recompiled the program. In my opinion, the volume in the new library is greatly exaggerated, which in the case of Internet stations can be manifested by large distortions at maximum volume. As you can see, at the standard level at which you generated the mp3 file, the sine wave is ok. However, the stations do not always control the volume level correctly. and a few stations have distorted volume.

BoloKonstruktor, 2023-07-12:
Increase the range with the 'setVolumeSteps' method until you get the same signal level with the 'setVolume( 21 )' value as in the older version of the library.

schreibfaul1, 2023-07-12:
Now I see what you mean. Digital filters are used, which not only attenuate the signal but can also amplify it. In the old version, the level was lower so that any amplification of the filters would not lead to distortion. With the new version, the overall level is lowered by the gain of the filters. Example: if the high pass is boosted by 6dB, the volume is lowered by 6dB beforehand. This ensures that the range from INT16_MIN ... INT16_MAX is not exceeded at 100% amplitude deviation.
I suspect that the output signal is now larger in amplitude (at the output of the DAC) than the connected amplifier can handle.

bhpjean, 2023-07-15:
No, it's not the amplifier's fault that it can't handle it. The amplifier has a volume control on the input. The problem is on the DAC side. he's not making it. I think the previous method was better, but that's just my opinion. In analog equipment, the bass and treble control does not work so that the overall volume is lowered and the timbre of the sound is boosted (boosted or lowered). in good audio equipment they do not use such methods because the sound is somewhat compressed and the signal level fluctuates. In my opinion, this is a bad method that degrades the sound. But you are the developer of the software and you decide it. Anyway, thank you for the great software and the time you spend creating it. I'll stay with the old method. For me it is better and the sound is much nicer, although it plays a bit quieter. It resembles and refers to the analog sound.

My internet tuners, which were created thanks to your software. Aren't they pretty?
Thank you
[image]

Arielhh, 2023-07-29:
I agree there is distortion in the sound in the new version 3.0.

schreibfaul1, 2023-07-29:
The old version ran at 15 bits/sample (50% level) because the digital filters can also amplify and there shouldn't be any overflow. The problem has now been solved and 16 bits/sample are the standard. The DAC should be able to do that, 
maybe there is a problem with the operating voltage

BoloKonstruktor, 2023-07-29:
Exactly, the normalized 16 bit signal is reproduced correctly without distortion. If, however, the audio stream has an abnormal level and in addition the filters are set to gain, the value of the coefficient 'm_corr' may not be sufficient to lower the level sufficiently to avoid distortion.

Arielhh, 2023-07-30:
> Exactly, the normalized 16 bit signal is reproduced correctly without distortion. If, however, the audio stream has an abnormal level and in addition the filters are set to gain, the value of the coefficient 'm_corr' may not be sufficient to lower the level sufficiently to avoid distortion.

Thanks :)

Arielhh, 2023-07-30:
> Exactly, the normalized 16 bit signal is reproduced correctly without distortion. If, however, the audio stream has an abnormal level and in addition the filters are set to gain, the value of the coefficient 'm_corr' may not be sufficient to lower the level sufficiently to avoid distortion.

> The old version ran at 15 bits/sample (50% level) because the digital filters can also amplify and there shouldn't be any overflow. The problem has now been solved and 16 bits/sample are the standard. The DAC should be able to do that, maybe there is a problem with the operating voltage

Thanks, there is no issue with the operating voltage. I tried better DAC chips but still the sound has some kind of distortion (not completely clear, which is a bit annoying).

pitlochry, 2023-08-02:
Thank you for your excellent work.

I can confirm that the sound is not completely clear (but I have no oscilloscope to observe it in more detail), maybe the "old" sound seemed a bit better. In my opinion best and most clear sound output is from the arduino-audio-tools lib from PSchatzmann. Totally subjective, of course.

But your lib is great! Thanks again.

Arielhh, 2023-08-06:
> Thanks, there is no issue with the operating voltage. I tried better DAC chips but still the sound has some kind of distortion (not completely clear, which is a bit annoying).

> The old version ran at 15 bits/sample (50% level) because the digital filters can also amplify and there shouldn't be any overflow. The problem has now been solved and 16 bits/sample are the standard. The DAC should be able to do that, maybe there is a problem with the operating voltage

I played with the setTone() function and managed to improve the sound to a minimum distortion level. Thanks again.

renatoianhez, 2023-09-26:
Hi! What tone combination did you find ideal? I have been using `audio.setTone(4, 0, 4);` and I notice that some stations with higher volume give distortion in the lower tones.

Arielhh, 2023-09-26:
> Hi! What tone combination did you find ideal? I have been using `audio.setTone(4, 0, 4);` and I notice that some stations with higher volume give distortion in the lower tones. Examples: https://italianradio.streamingmedia.it/play https://27343.live.streamtheworld.com/RADIO_ALVORADAAAC.aac

I used audio.setTone(0, 6, 4); for a small 4cm speaker but after further thinking I replaced the DAC with PCM5102 and the sound is great now.
If you are using the MAX chip consider replacing it as the DAC in this chip is not of good quality.

renatoianhez, 2023-09-28:
I use PCM5102 exactly, with a PAM8403 amp, 2'' speakers, stereo. [This is my radio](https://github.com/renatoianhez/Lolin32-radio).

renatoianhez, 2023-09-29:
> I'll test this 0,6,4... I use PCM5102 exactly, with a PAM8403 amp, 2'' speakers, stereo. [This is my radio](https://github.com/renatoianhez/Lolin32-radio).

Its combination greatly reduced the noise!

Arielhh, 2023-09-29:
With the PCM DAC you should not have a problem. I guess that the issue you
experience will not be fixed by playing with the tone values.  In my
opinion you should look somewhere else. See that project
https://github.com/Arielhh/ESP32-Radio-Internet

renatoianhez, 2023-10-01:
> With the PCM DAC you should not have a problem. I guess that the issue you experience will not be fixed by playing with the tone values. In my opinion you should look somewhere else. See that project https://github.com/Arielhh/ESP32-Radio-Internet
> […](#)

I noticed that the SPI mode option was in DIO, and not QIO as I usually do. Is there any influence?
[image]

closed 2024-01-24 by github-actions[bot]