schreibfaul1/ESP32-audioI2S #569: Incorrect mix up left/right channel

BoloKonstruktor, 2023-07-28:
Library modification (mix up left/right channel) introduced incorrect playback of left and right channels. The left channel plays in the right and the right in the left. This can be checked with any stereo test.

BoloKonstruktor, 2023-08-05:
The simplest solution to the problem:
```
// ...
//"przelutowanie" kanałów
int16_t r  = sample[LEFTCHANNEL];
int16_t l = sample[RIGHTCHANNEL];
sample[LEFTCHANNEL] = l;
sample[RIGHTCHANNEL] = r;
	
computeVUlevel(sample);
// ...
```
The only question is why the Author changed channels in Commits on July 4, 2023?

schreibfaul1, 2023-08-05:
was removed with the last pull request
I had a swap in the audio library and have now swapped it back in the application.

Xk-w, 2023-08-17:
> Library modification (mix up left/right channel) introduced incorrect playback of left and right channels. The left channel plays in the right and the right in the left. This can be checked with any stereo test.

Yes, I also encountered this issue on July 1, 2023

schreibfaul1, 2023-08-17:
really?
[l_r.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/12365487/l_r.zip)

closed 2023-08-17 by schreibfaul1

BoloKonstruktor, 2023-08-17:
This test [test_stereo.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/12369233/test_stereo.zip) (publicly available on the internet) plays correctly on all my media devices except for a player using the 'ESP32-audioI2S' library version July 1, 2023.

The problem is already present in the displayed VU-meter, so I exclude incorrect connection of the DAC to the amplifier.