schreibfaul1/ESP32-audioI2S #570: Music files stop before actual end

kiwi85, 2023-08-02:
I am using wrover chip with psram and Arduino ide. Everything works fine despite any mp3 file stops about 1 s before end. Also loop takes quite a while to restart the file.
Any ideas on how to address this issue ?

schreibfaul1, 2023-08-04:
I have a test file here, 1KHz, 80% 0.8s pause, 0.2s tone, 5 times.
This works as expected.
[image]
[5pulse.wav.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/12257479/5pulse.wav.zip)

lombyp, 2024-01-01:
Hello and congratulations for your work, I'm using wrover **https://www.makerfabs.com/eps32-audio-player.html** and Arduino IDE, I'm having a problem playing .wav files, every time I use the function **audio.setAudioPlayPosition(int);**  the playback of the song stops before its actual end, sometimes even halfway.
Have you already noticed this problem?
Can I call the function while the song is playing or do I have to pause it first?
to be honest I've tried them all but the problem still occurs.

schreibfaul1, 2024-01-02:
Hello @lombyp,
I was able to simulate the error, especially with PSRAM and short files. Hope it works now.

lombyp, 2024-01-02:
It works now! many thanks

closed 2024-02-16 by github-actions[bot]