schreibfaul1/ESP32-audioI2S #578: Crash while decoding Ogg Vorbis file with long Type 5 header

ma261065, 2023-08-27:
Similar to issue https://github.com/schreibfaul1/ESP32-audioI2S/issues/558, it seems that there are some files out there which have overly long segments.

Example https://archive.org/download/gd1989-07-19.fob.nak300.holtz.jaz.106492.flac16/GD1989-07-19trk01.ogg

Dumping that file shows that Ogg page 2, which contains the Type 5 Vorbis header starting at 0x42a should finish at 0x12fd, but the next Ogg page actually starts at 0x1400

It seems that the fix in https://github.com/schreibfaul1/ESP32-audioI2S/issues/558 doesn't work in this case, and the decoder crashes with a Guru Meditation Error.

Is it possible to extend the fix in the other issue to handle this case?

schreibfaul1, 2023-08-28:
Hello @ma261065,
thank you for your report. I found out that the codebooks are not in the usual place. The first step is now to avoid a crash. So far I don't have a solution for the problem. I will take care of it in the near future.

schreibfaul1, 2023-08-28:
I hope I have fixed the problem without adding new errors.

ma261065, 2023-08-28:
Thank you for making the player more robust, even in the face of technically invalid files.

It's annoying to have to deal with these, but from a user perspective they just expect all files to play, so this is worthwhile work.

I note that Windows, Google & Linux players all deal with these weird files, so I guess they also hit this problem at some point in their development.

I will test and see if I see any other issues.

closed 2024-01-22 by github-actions[bot]