schreibfaul1/ESP32-audioI2S #579: No MCLK signal generated on the ESP32-S3

Thecactusman0, 2023-08-28:
I can't get an MCLK signal out of my ESP32-S3, as seen in my code below I originally had it on gpio46 but I saw in audio.cpp and the wiki that only gpio 0,1,3 are supported. Ive tested with each of these and I can find no clock signal on the pins with my oscilloscope. 

```
#include "Audio.h"
#include "ES8327.h"

#define I2S_DOUT      48
#define I2S_BCLK      45
#define I2S_LRC       47
#define I2S_MCLK      46

ES8327 codec(Wire,18,0x18);//wire object,hp_int, address
Audio audio;

String ssid =     "";
String password = "";

void setup()
{
delay(5000);

Serial.println("Starting");
Wire.begin(15,16);
Serial.begin(115200);

WiFi.disconnect();
WiFi.mode(WIFI_STA);
WiFi.begin(ssid.c_str(), password.c_str());
while (WiFi.status() != WL_CONNECTED) delay(1500);

 if (!codec.begin()) 
 {
    Serial.println("Failed to find Codec");
    while (1) 
    {
      delay(10); //stop code in its tracks
    }
 }
  Serial.println("Codec Found!");
  
  codec.setWordLength(16);
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.i2s_mclk_pin_select(I2S_MCLK);
  audio.setVolume(5); // 0...21
  
  audio.connecttohost("https://github.com/schreibfaul1/ESP32-audioI2S/raw/master/additional_info/Testfiles/Olsen-Banden.mp3");        // mp3
}

void loop() 
{
audio.loop();
}

void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
```

The ES8327 code is just for the configuration of the codec ic, this code works fine on the standard esp32 and other audio libraries have been able to generate the mclk signal.

schreibfaul1, 2023-08-28:
The restrictions only apply to the ESP32 and only there `i2s_mclk_pin_select` is required. For the ESP32-S3 `setPinout(uint8_t BCLK, uint8_t LRC, uint8_t DOUT, int8_t DIN, int8_t MCK)` is sufficient.

Thecactusman0, 2023-08-30:
Thanks for the help

closed 2023-08-30 by Thecactusman0