schreibfaul1/ESP32-audioI2S #586: ESP32 reboot loop if function not defined

speedykris, 2023-09-15:
I fought with an issue for few days. Even with the most basic web stream player code on WEMOS ESP32 LOLIN Lite the code was causing reboot with Core 1 panic just after trying to connect to the stream.
It occurred that the problem was caused by missing optional part in the end:

void audio_info(const char *info){
    Serial.println(info);
}

If this is left undefined there is some kind of a problem with the buffer. I'm not a C programmer but it was explained to me in such way (there may be mistakes but it's just a translation from my native language and I don't understand this even in my language :D):
There's something about the buffer problems due to the function defined as "weak" in the library. If it's empty, it works, but if it is not defined there are some buffer issues. It has something to do with a using or not a debug in the library.
Hopefully it's helpful.

schreibfaul1, 2023-09-16:
I cannot understand this. If audio_info() does not exist, the call is not started. Do you have an example where the error occurs?

speedykris, 2023-09-16:
#include "Audio.h"
#include "WiFi.h"
#include "Arduino.h"

#define I2S_DOUT 25
#define I2S_BCLK 27
#define I2S_LRC 26

String ssid = "*********";
String password = "*******";

Audio audio;

void setup(){
  Serial.begin(115200);
  WiFi.disconnect();
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(),password.c_str());
  Serial.println("Connected");
  Serial.println(WiFi.localIP());
  Serial.print("RRSI: ");
  Serial.println(WiFi.RSSI());
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  audio.setVolume(15);
  audio.connecttohost("http://stream.open.fm/378?type=.aac");  
  Serial.println ("Connect do stream");
}
void loop()
{
  audio.loop();
}

The above code compiles and loads without issues to WEMOS Lolin32 Lite board. But enters infinite reboot loop when going to audio.connecttohost line with Core 1 panic:
[image]

rboeije1, 2023-09-18:
Not sure if the presence of that function is the real cause. However, I have (not) exactly the same by that means, that I have the same reboot loop if I run an axample, like 'I2Saudio'. I would say one of the simplest examples. 
2 notes:
- A week or 3 ago it compiled fine and ran. (In fact the somewhat more extended demo with local storage files, but as my modified version did not run anymore, I tried the original examples)
- On the front page it mentions it needs S3, but I ran it successfully on the old pico kit V4.1 or esp32_devkit_V4 a few weeks ago. Now it reboots after the connecttohost().

The date of the files of the library on my PC are from 23-8-2023. As far as I can see nothing changed on this library in between on my PC. So I am confused too. Something changed on the ESP32 core? When I first downloaded I was so happy that it all worked out of the box...

schreibfaul1, 2023-09-18:
You probably have the Arduino version 2.0.12 or 2.0.13. This will not run the audioI2S version from August. You can either go back to 2.0.11 or update to the current master.

speedykris, 2023-09-18:
I'm on newest Arduino IDE (2.2.1). This is the only version I installed.

Wysłano z programu Outlook dla systemu Android<https://aka.ms/AAb9ysg>

schreibfaul1, 2023-09-18:
I did not mean the version of the Arduino IDE, but the version [arduino-esp32](https://github.com/espressif/arduino-esp32)
[image]

Bored0ne, 2023-09-22:
This was it. I had this as my platform_packages.
```
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```
Removed it, got a fresh build going, and kernel dump begone

schreibfaul1, 2023-09-22:
You could do it like this, then the correct versions and the current master are fetched:
````
[env]
platform = https://github.com/platformio/platform-espressif32.git#v6.4.0 ; ESP-IDF v4.4.5
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.13
lib_deps = https://github.com/schreibfaul1/ESP32-audioI2S.git
````

closed 2024-01-21 by github-actions[bot]