schreibfaul1/ESP32-audioI2S #587: Generic Error

Maverik1408, 2023-09-15:
Hi everyone, until yesterday my code worked, now it keeps crashing, can you help me?

`
#include "Audio.h"
#include <WiFi.h>

Audio audio(true, I2S_DAC_CHANNEL_BOTH_EN);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.setDebugOutput(true);
WiFi.setAutoReconnect(true);
WiFi.begin("xxxx", "xxxx");
      Serial.println("Connection WiFi..");
      while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
        
      }
      Serial.println("");
      Serial.print("Connected to WiFi network with IP Address: ");
      Serial.println(WiFi.localIP().toString());

Serial.println("1");
  audio.setVolume(12);  // 0...21
Serial.println("2");
audio.connecttospeech("WiFi Connected", "en");
  //audio.connecttohost("http://stream.dancewave.online:8080/dance.mp3");  //  128k mp3
  Serial.println("3");
}

void loop() {
  // put your main code here, to run repeatedly:
  audio.loop();
}

`

this is the backtrace

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ef7af  PS      : 0x00060130  A0      : 0x800ef8b8  A1      : 0x3ffb2160  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x0000002d  A5      : 0x3ffb1ff0  
A6      : 0x0000002d  A7      : 0x3ffb1ff0  A8      : 0x801780d2  A9      : 0x3ffb1db0  
A10     : 0x00000000  A11     : 0x3ffb20f0  A12     : 0x00000006  A13     : 0x00001000  
A14     : 0x00000006  A15     : 0x3ffb2014  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x40089bb9  LEND    : 0x40089bc9  LCOUNT  : 0xfffffffe  

Backtrace: 0x400ef7ac:0x3ffb2160 0x400ef8b5:0x3ffb2190 0x400db9d0:0x3ffb21b0 0x400ddf63:0x3ffb21d0 0x400d2ea7:0x3ffb2240 0x400f48ae:0x3ffb2290

PC: 0x400ef7af: WiFiClientRxBuffer::r_available() at /Users/ChicchiDani/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/src/WiFiClient.cpp:46
EXCVADDR: 0x00000010

Decoding stack results
0x400ef7ac: WiFiClientRxBuffer::r_available() at /Users/ChicchiDani/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/src/WiFiClient.cpp:44
0x400ef8b5: WiFiClient::flush() at /Users/ChicchiDani/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi/src/WiFiClient.cpp:158
0x400db9d0: Audio::setDefaults() at /Users/ChicchiDani/Documents/Arduino/libraries/ESP32-audioI2S/src/Audio.cpp:335
0x400ddf63:  is in Audio::connecttospeech(char const*, char const*) (/Users/ChicchiDani/Documents/Arduino/libraries/ESP32-audioI2S/src/Audio.cpp:784).
0x400d2ea7: setup() at /Users/ChicchiDani/Documents/Arduino/sketch_sep15a/sketch_sep15a.ino:26
0x400f48ae: loopTask(void*) at /Users/ChicchiDani/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/cores/esp32/main.cpp:42

schreibfaul1, 2023-09-16:
This is probably related to the new Arduino version 2.0.12. A PR was incorporated for this and the bug was fixed.

Maverik1408, 2023-09-16:
I'm sorry, I didn't understand.. do I have to change the version of Arduino IDE? I have 2.2.1.. thanks

schreibfaul1, 2023-09-16:
I didn't mean the version of the Arduino IDE. You can find the installed Arduino version for the ESP32 in the Boards Manager if you enter "Arduino ESP32".
 [image]

I think you have version 2.0.12 installed. There is a patch for this version, please try it with the latest version of my library.

Maverik1408, 2023-09-16:
Thank you i downgrade ti 2.0.9 and work!!!

closed 2023-09-16 by Maverik1408