schreibfaul1/ESP32-audioI2S #593: statical-code-check with Platform-IO

tueddy, 2023-09-24:
Hi Wolle,

nothing critical, just noticed a few hints/warnings while exec statical-code-check with Platform-IO:

`src\AudioPlayer.cpp:917: [low:style] Variable 'tmp' can be declared as pointer to const [constVariablePointer]
src\AudioPlayer.cpp:1086: [low:style] Variable 'last' can be declared as pointer to const [constVariablePointer]
src\AudioPlayer.cpp:1087: [low:style] Variable 'first' can be declared as pointer to const [constVariablePointer]
.pio\libdeps\lolin_d32_pro_sdmmc_pe\ESP32-audioI2S-master\src\Audio.h:389: [medium:warning] Uninitialized variable: result [uninitvar]
.pio\libdeps\lolin_d32_pro_sdmmc_pe\ESP32-audioI2S-master\src\Audio.h:410: [low:style] Variable 'len' is assigned a value that is never used. [unreadVariable]
src\AudioPlayer.cpp:1137: [low:style] Variable 'swap' is assigned a value that is never used. [unreadVariable]`

check is here:

[image]

greetings
@tueddy

schreibfaul1, 2023-09-25:
Hi tueddy,
hope the test-unit doesn't find anything now :-)

tueddy, 2023-09-25:
Perfect & lightning fast, Thank's!

closed 2023-09-25 by tueddy