schreibfaul1/ESP32-audioI2S #601: Format .m3u8 not working

tv4you2016, 2023-10-10:
hello @schreibfaul1

When I try to reproduce this url the reboot esp

Media URL: https://stream-hls.bauermedia.pt/comercial.aac/playlist.m3u8

Log: 

> [ 66660][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
> [ 66661][I][Audio.cpp:5861] ts_parsePacket(): parseTS reset
> [ 66662][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 148232
> [ 66669][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
> [ 66834][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
> [ 66836][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
> [ 66839][D][ssl_client.cpp:179] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> [ 66847][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
> [ 66856][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
> [ 68066][V][ssl_client.cpp:293] start_ssl_client(): Verifying peer X.509 certificate...
> [ 68066][V][ssl_client.cpp:301] start_ssl_client(): Certificate verified.
> [ 68069][V][ssl_client.cpp:316] start_ssl_client(): Free internal heap after TLS 105484
> [ 68077][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 154 bytes...
> [ 68478][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: HTTP/1.1 200 OK
> [ 68479][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: content-length: 216
> [ 68484][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: pragma: no-cache
> [ 68492][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: expires: Mon, 26 Jul 1997 05:00:00                                                                                                                                GMT
> [ 68502][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: instance-id: 75831741d5e5a758a441bd                                                                                                                               93c71b9fcb
> [ 68513][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: server: edge-bauerpthls-08-thn.shar                                                                                                                               p-stream.com 9.0.7
> [ 68524][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: accept-ranges: bytes
> [ 68532][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: content-type: application/vnd.apple                                                                                                                               .mpegurl
> [ 68543][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: cache-control: no-cache
> [ 68552][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: set-cookie: ListenerInfo=reset; Pat                                                                                                                               h=/; Domain=stream-hls.bauermedia.pt; Expires=Tue, 10 Oct 2023 19:09:49 GMT; SameSite=None; Secure
> [ 68570][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: x-response-server: edge-bauerpthls-                                                                                                                               08
> [ 68580][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: access-control-allow-origin:
> [ 68589][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: access-control-allow-credentials: t                                                                                                                               rue
> [ 68599][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: set-cookie: SERVERID=edge-bauerpthl                                                                                                                               s-08; path=/; Secure; SameSite=None
> [ 68611][I][Audio.cpp:4072] parseHttpResponseHeader(): now parse playlist
> [ 68641][E][Audio.cpp:2640] readPlayListData(): m3u8 playlists requires PSRAM enabled!
> [ 68646][I][Audio.cpp:2735] readPlayListData(): pl=0 "#EXTM3U"
> [ 68650][I][Audio.cpp:2735] readPlayListData(): pl=1 "#EXT-X-STREAM-INF:BANDWIDTH=128000,CODECS="mp4a.40.2""
> [ 68660][I][Audio.cpp:2735] readPlayListData(): pl=2 "https://stream-hls.bauermedia.pt:443/comercial.aac/playl                                                                                                                               ist.m3u8?listeningSessionID=650047c5723f9ac7_2451400_0H2GS9rM__00000003TZj&downloadSessionID=0"
> [ 68689][I][Audio.cpp:2943] parsePlaylist_M3U8(): redirect https://stream-hls.bauermedia.pt:443/comercial.aac/                                                                                                                               playlist.m3u8?listeningSessionID=650047c5723f9ac7_2451400_0H2GS9rM__00000003TZj&downloadSessionID=0
> [ 68705][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
> [ 68707][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 147708
> [ 68714][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
> [ 68767][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
> [ 68768][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
> [ 68771][D][ssl_client.cpp:179] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> [ 68780][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
> [ 68788][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
> [ 70130][V][ssl_client.cpp:293] start_ssl_client(): Verifying peer X.509 certificate...
> [ 70130][V][ssl_client.cpp:301] start_ssl_client(): Certificate verified.
> [ 70133][V][ssl_client.cpp:316] start_ssl_client(): Free internal heap after TLS 105328
> [ 70141][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 224 bytes...
> [ 71049][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: HTTP/1.1 200 OK
> [ 71050][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: content-length: 3381
> [ 71055][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: pragma: no-cache
> [ 71063][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: expires: Mon, 26 Jul 1997 05:00:00                                                                                                                                GMT
> [ 71073][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: instance-id: ded8f868bfaa7d54552356                                                                                                                               5778f5d113
> [ 71083][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: server: edge-bauerpthls-09-gos2.sha                                                                                                                               rp-stream.com 9.0.7
> [ 71095][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: accept-ranges: bytes
> [ 71103][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: content-type: application/vnd.apple                                                                                                                               .mpegurl
> [ 71114][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: cache-control: no-cache
> [ 71123][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: set-cookie: ListenerInfo=ZGVkOGY4Nj                                                                                                                               hiZmFhN2Q1NDU1MjM1NjU3NzhmNWQxMTM7MDs7MDtnZW5lcmF0ZWRfYmY2M2YwMTBjMTgxNmM3YWViZjAxYjA0NGQ1NDA5ODA7MTY5Njk2ODQ5                                                                                                                               ODk4NDswLzEwMDA7ODAwMDA7dCwxNjk2OTY4NDc4OTc0O3QsMTY5Njk2ODQ4ODk3ODt0LDE2OTY5Njg0OTg5ODQ=; Path=/; Domain=strea                                                                                                                               m-hls.bauermedia.pt; Max-Age=40; Expires=Tue, 10 Oct 2023 20:10:32 GMT; SameSite=None; Secure
> [ 71160][I][Audio.cpp:3866] parseHttpResponseHeader(): httpResponseHeader: set-cookie: AISSessionId=0_650047c5                                                                                                   [... cut ...] 3259:0x3ffb2820
> 
> 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...
> ets Jul 29 2019 12:21:46
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13132
> load:0x40080400,len:3036
> entry 0x400805e4
> [     6][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> 
> 
>

schreibfaul1, 2023-10-11:
Hi @tv4you2016,
this is the first m3u8 playlist I've seen in xml format. The crash comes from the large line length and the buffer limit overrides. Additionally, there is the problem that the chunks are not always played in the correct order. This will take some time.

schreibfaul1, 2023-10-16:
I think I've got a handle on it now. I get the correct order of the chunks via the media sequence number. https://stream-hls.bauermedia.pt/comercial.aac/playlist.m3u8 has been running without any problems for an hour.
The changes are not in the master. The branch "Arduino V3.0" (with IDF 5.1) contains the relevant updates. As soon as there is a stable Arduino 3 version, the branch will become the master.

Best regards from Berlin
Wolle

schreibfaul1, 2023-11-29:
This error will certainly not occur again. Whereby the station `chunks-bauerpt.sharp-stream.com` is difficult to receive for me, ping times are okay, but 25% data loss.

closed 2025-02-12 by github-actions[bot]