schreibfaul1/ESP32-audioI2S #603: SPIClass cause SW_CPU_RESET

CNawalol, 2023-10-22:
When I initialize the SD card with SPIClass, audioI2s does not play MP3s. If I use SPI, it reads fine.

code:
```
//
// Created by awalol on 2023/10/15.
//
#include <SPI.h>
#include <SD.h>
#include "Arduino.h"
#include "Audio.h"

#define SD_SPI_SCK 32
#define SD_SPI_MISO 33
#define SD_SPI_MOSI 25
#define SD_SPI_CS 15
#define I2S_DOUT 14
#define I2S_LRCK 27
#define I2S_BCK 12

Audio audio;

void setup() {
    Serial.begin(115200);

    // can't work
    SPIClass H(HSPI);
    H.begin(SD_SPI_SCK, SD_SPI_MISO, SD_SPI_MOSI, SD_SPI_CS);
    if (!SD.begin(SD_SPI_CS, H)) {
        Serial.println("SD 读取失败");
    }

    // can work
    /*SPI.begin(SD_SPI_SCK, SD_SPI_MISO, SD_SPI_MOSI,SD_SPI_CS);
    if (!SD.begin(SD_SPI_CS,SPI)) {
        Serial.println("SD 读取失败");
    }*/

    // 初始化Audio
    File root = SD.open("/");

    File file = root.openNextFile();
    while (file) {
        Serial.println(file.path());
        file = root.openNextFile();
    }

    audio.setPinout(I2S_BCK, I2S_LRCK, I2S_DOUT);

    audio.connecttoSD("test.mp3");
    audio.setVolume(12);
}

void loop() {
    audio.loop();
}
```

Output when using SPIClass:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
/System Volume Information
/test.txt
/foo.txt
/test.mp3
/music
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e745f  PS      : 0x00060e30  A0      : 0x800d3242  A1      : 0x3ffd1400  
A2      : 0x0000000e  A3      : 0x000000ff  A4      : 0x00000086  A5      : 0x000007c0  
A6      : 0x3ffd3898  A7      : 0x00000000  A8      : 0x0000000e  A9      : 0x00000000  
A10     : 0x00820000  A11     : 0x07c00000  A12     : 0x3ffd3898  A13     : 0x00060e23  
A14     : 0x00060e20  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000e  LBEG    : 0x400845a1  LEND    : 0x400845a9  LCOUNT  : 0x00000027  

Backtrace: 0x400e745c:0x3ffd1400 0x400d323f:0x3ffd1420 0x400d3741:0x3ffd1440 0x400d37b1:0x3ffd1460 0x400d3b11:0x3ffd1490 0x400d3b3e:0x3ffd14b0 0x400f679e:0x3ffd14d0 0x400f722e:0x3ffd14f0 0x400f8282:0x3ffd1510 0x400f9fc7:0x3ffd1550 0x400f3532:0x3ffd1580 0x4008707d:0x3ffd15a0 0x40086fc1:0x3ffd15c0 0x40130ed6:0x3ffd15e0 0x40130f69:0x3ffd1620 0x4012bb87:0x3ffd1640 0x400d33ff:0x3ffd1660 0x400d9946:0x3ffd1680 0x400dc99a:0x3ffd16b0 0x400d2d9a:0x3ffd16d0 0x400e8321:0x3ffd16f0

ELF file SHA256: e1e193aec4b60a30

Rebooting...
ets Jul 29 2019 12:21:46
```

Thanks for your help.

CNawalol, 2023-10-22:
add audio_info output:
```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
/System Volume Information
/test.txt
/foo.txt
/test.mp3
/music
PSRAM not found, inputBufferSize: 6399 bytes
buffers freed, free Heap: 217300 bytes
Reading file: "/test.mp3"
MP3Decoder has been initialized, free Heap: 189584 bytes
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e72df  PS      : 0x00060e30  A0      : 0x800d3252  A1      : 0x3ffd1400  
A2      : 0x0000000e  A3      : 0x000000ff  A4      : 0x00000088  A5      : 0x00006d80  
A6      : 0x3ffd3898  A7      : 0x00000000  A8      : 0x0000000e  A9      : 0x00000000  
A10     : 0x00850000  A11     : 0x6d800000  A12     : 0x3ffd3898  A13     : 0x00060e23  
A14     : 0x00060e20  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000e  LBEG    : 0x400845a1  LEND    : 0x400845a9  LCOUNT  : 0x00000027  

Backtrace: 0x400e72dc:0x3ffd1400 0x400d324f:0x3ffd1420 0x400d3751:0x3ffd1440 0x400d37c1:0x3ffd1460 0x400d3b21:0x3ffd1490 0x400d3b4e:0x3ffd14b0 0x400f661e:0x3ffd14d0 0x400f70ae:0x3ffd14f0 0x400f8102:0x3ffd1510 0x400f9e47:0x3ffd1550 0x400f33b2:0x3ffd1580 0x4008707d:0x3ffd15a0 0x40086fc1:0x3ffd15c0 0x40130d56:0x3ffd15e0 0x40130de9:0x3ffd1620 0x4012ba07:0x3ffd1640 0x400d340f:0x3ffd1660 0x400d9856:0x3ffd1680 0x400dc836:0x3ffd16b0 0x400d2d9a:0x3ffd16d0 0x400e81a1:0x3ffd16f0

ELF file SHA256: 5dcf2bf75851d6f2
```

schreibfaul1, 2023-10-22:
Hello,

`SPIClass H(HSPI);` must be global, must not be in setup()

CNawalol, 2023-10-28:
Thanks! Works great.

closed 2023-10-28 by CNawalol