schreibfaul1/ESP32-audioI2S #606: Wrong ID3 data for "Tausendfüßler"

tueddy, 2023-11-08:
Hi Wolle,

In general the ID3 tag information works very well but in this case i get wrong result. 
Expected:

Album: "Eule findet den Beat - mit Gefühl (Musik-Hörspiel)"
Title: "Der Tausendfüßler und die Ukulele"

Got:

...
I [38680] info        : ID3 normal frames
I [38842] id3data     : SettingsForEncoding: Lavf58.29.100
I [39002] id3data     : Alb
I [39161] id3data     : Artist: Eule
I [39324] id3data     : `Tit`
...

Encoding seems to be 0=ASCII, maybe a problem with "üß"?

[Tausendfüßler.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/13301644/Tausendfussler.zip)

schreibfaul1, 2023-11-11:
Hi tueddy, 
the error occurs when the metadata is neither Uincode nor pure ASCII. In this case, the ID3 data is in ISO_8859-1 format. I have now taken this into account. 

best regards
Wolle

tueddy, 2023-11-12:
Work's like a charme, thank's for the very fast fix!

closed 2023-11-12 by tueddy

tueddy, 2023-11-13:
Sorry to reopen but i found some kind of regression.
Attached MP3 was fine before but with this fix it shows wrong ID3 data (ü-Umlaut)

Before this fix:
`id3data     : Title:  findet den Beat - mit Gefühl`

Now:
`id3data     : Title:  findet den Beat - mit Gef `

[Eule - findet den Beat - mit Gefühl.zip](https://github.com/schreibfaul1/ESP32-audioI2S/files/13333927/Eule.-.findet.den.Beat.-.mit.Gefuhl.zip)

schreibfaul1, 2023-11-13:
That looks quite good to me
 [image]
 In the metadata there are still control characters like carriage return, I filtered them all out, but maybe I missed something?

tueddy, 2023-11-13:
Sorry, i forget to make a clean + full build after update ;-(
Everything works as expected now, thanks!