schreibfaul1/ESP32-audioI2S #607: connecttospeech followed by connecttohost doesent work?

stieneker, 2023-11-10:
Hello, 
i would like to use the i2s libary von Schreibfaul to ensure that the text is announced, but then a web radio station is received (example) (here in setup, same problem in loop)?
Why does this not work?

  // Connect MAX98357 I2S Amplifier Module
  audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
  // Set the volume
  audio.setVolume(12);
  audio.connecttospeech("Internet raadio", "de");
  Serial.println("hier sollte sie geredet haben? ");
  
  audio.setVolume(10);
  // audio.connecttohost(stations[StZaehler].c_str());
  audio.connecttohost("http://streams.80s80s.de/web/mp3-192/streams.80s80s.de/");

By the way, what is?
// audio.connecttomarytts("Hier ist Mary","de",HIGH);
//audio.connecttomarytts()

Greetings Manfred

schreibfaul1, 2023-11-10:
you wait until connecttospeech(.....) is finished. Then the event audio_eof_speech(const char *info) is called. In the simplest case, you can use this to set up another connection.

void setup(){ 
    ... 
    audio.connecttospeech("Internet raadio", "en");
     ...
 }

void loop(){ 
    audio.loop();
}

void audio_eof_speech(const char *info){
     audio.connecttohost("http://streams.80s80s.de/web/mp3-192/streams.80s80s.de/");
}

Mary is a speech synthesis with poor quality. It is no longer included in the new versions.

stieneker, 2023-11-13:
Oh, thanks a lot for this infos.
Sometimes you get a knot in your Head.
Greetings Manfred (DB3BP)

closed 2025-02-12 by github-actions[bot]