schreibfaul1/ESP32-audioI2S #609: Play Raw OPUS.

elshnkhll, 2023-11-13:
Thanks for GREAT work and countless hours invested into this unique project.

I am receiving raw OPUS packets (~120 byte) on a port of ESP32 every 20ms. Packets are not wrapped (no MP3 or OGG). First byte is F4 = 11110100, meaning they are in CELT mode, stereo and 1 frame per packet, 48000 sample rate. Is it feasible to adapt your code to play this kind of stream?

Thanks in advance.
With wishes of success.

schreibfaul1, 2023-11-13:
I don't know that exactly. Do you have an example of such a file?

elshnkhll, 2023-11-25:
It is not a file. I am getting UDP packets delivered to port:

`
F4 = 11110 1 00
     CELT  ^  ^ - 1 Frame in the packet
     mode  | 
           Stereo

0010                     f4 07 1b 7c e7 ec a4 5f ce 40
0020   90 38 c2 7e 5d e0 99 9c 73 bc 45 8b fb 45 56 fb
0030   6b 8d d5 df 30 b2 bb a1 dd a5 9b c1 5a 68 ae 0c
0040   f5 97 0c d0 3f 2e ec f7 f1 54 90 c3 c7 7b 71 ef
0050   c9 ec 53 d2 e0 58 91 8d 85 60 55 f7 92 35 99 fe
0060   a0 d3 0c 52 f7 ab 4e 19 20 01 ed 1b c5 f4 03 1b
0070   3f b3 7c 97 8b 9e d2 36 d2 cf ce 43 e2 d7 9e 58
0080   f3 4b 05 a4 14 01 b7 32 19 ad
`

elshnkhll, 2023-11-25:
It is not a file. I am getting UDP packets delivered to a port of ESP32. Here is a payload example extracted from the event.data.

0010:                   f4 07 1b 7c e7 ec a4 5f ce 40
0020: 90 38 c2 7e 5d e0 99 9c 73 bc 45 8b fb 45 56 fb
0030: 6b 8d d5 df 30 b2 bb a1 dd a5 9b c1 5a 68 ae 0c
0040: f5 97 0c d0 3f 2e ec f7 f1 54 90 c3 c7 7b 71 ef
0050: c9 ec 53 d2 e0 58 91 8d 85 60 55 f7 92 35 99 fe
0060: a0 d3 0c 52 f7 ab 4e 19 20 01 ed 1b c5 f4 03 1b
0070: 3f b3 7c 97 8b 9e d2 36 d2 cf ce 43 e2 d7 9e 58
0080: f3 4b 05 a4 14 01 b7 32 19 ad

closed 2025-03-28 by github-actions[bot]