schreibfaul1/ESP32-audioI2S #610: RTOS Multi-Task Program: Issues with Audio Playback and Custom File Handling

AmitPadhiyarOfficial, 2023-11-14:
I am working on an RTOS multi-task program where an audio file runs on task 1, core 0, and my text file runs on task 2, core 1. Essentially, task 1 plays audio and stores the current time in a variable. Meanwhile, task 2, where my custom file format is, receives the time from the current time. It has a converter that changes the time to file position, seeks the file position, retrieves the data, and must manage a playlist of multiple audio and custom text format files. For the first two times, the audio plays fine, and the custom file provides data properly. Unfortunately, on the third attempt, I believe I'm seeing random data, and my audio files are also not working properly.

**Simple.ino**

```cpp
#include "Audio.h"

enum States { selectData,
              selectAudio,
              playing };

int acount = 1;
int dcount = 1;

Audio afile(true);
File dfile;
States state = selectData;
uint32_t currentTime;

void setup() {
  Serial.begin(115200);
  delay(2000);

  Serial.println("PSP started.");

  if (!SD.begin(5)) {
    Serial.println("SD failed.");
    return;
  }

  afile.setVolume(afile.maxVolume());

  xTaskCreatePinnedToCore(
    audioTaskSource,   /* Task function. */
    "audioTaskSource", /* name of task. */
    10000,             /* Stack size of task */
    NULL,              /* parameter of the task */
    1,                 /* priority of the task */
    NULL,              /* Task handle to keep track of created task */
    0);                /* pin task to core 0 */

  xTaskCreatePinnedToCore(
    dataTaskSource,   /* Task function. */
    "dataTaskSource", /* name of task. */
    10000,            /* Stack size of task */
    NULL,             /* parameter of the task */
    1,                /* priority of the task */
    NULL,             /* Task handle to keep track of created task */
    1);               /* pin task to core 1 */
}
void loop() {
}
void audioTaskSource(void *pvParameters) {
  while (1) {
    audioManager();
    vTaskDelay(0);
  }
}
void dataTaskSource(void *pvParameters) {
  while (1) {
    dataManager();
    vTaskDelay(30);
  }
}

void audioManager() {
  if (state == selectAudio) {
    if (afile.connecttoFS(SD, "Demo.mp3")) {
      Serial.printf("Audio opend %d.\n", (acount++));
    } else {
      Serial.printf("Audio not opend %d.\n", (acount++));
    }
    state = playing;
  } else if (state == playing) {
    afile.loop();
    currentTime = afile.getAudioCurrentTime();

    if (!afile.isRunning()) {
      dfile.close();
      afile.stopSong();
      state = selectData;
    }
  }
}
void dataManager() {
  if (state == selectData) {
    dfile = SD.open("/simple.txt");
    if (dfile) {
      Serial.printf("Data opend %d.\n", (dcount++));
    } else {
      Serial.printf("Data not opend %d.\n", (dcount++));
    }
    state = selectAudio;
  } else if (state == playing) {
    //dfile.seek(currentTimeToDataPosConverter(currentTime));

    if (dfile.available()) {
      int databyte = dfile.read();
      Serial.print("Databyte: ");
      Serial.println(databyte);
    }

    if (!dfile.available()) {
      dfile.close();
    }
  }
}
```

**Simple.txt**
```
1ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789|2ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789|3ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789
```

**Task Steps**
- The default state is selectData.
- In dataManager(), during the selectData state, open the txt file, and set the state to selectAudio.
- After transitioning to the selectAudio state, audioManager() opens the audio file using connecttoFS() and sets the state to playing.
- Both functions have a playing state. audioManager() runs the audio loop, sets the current time into a variable, and dataManager() retrieves the current time, converts it into a file position. The text file position is set, and data is read in bytes.
- If no audio data is available, it calls the file's close() and audio's stopSong() functions, then sets the state back to selectData. Essentially, it repeats the first step.

**Error**
After the second attempt, the third attempt produces random data from the 'Simple.txt' file, and the audio is not playing well.

schreibfaul1, 2023-11-14:
At first glance, I would say that message queues should be set up between the tasks to avoid collisions. If different tasks use the same "critical" sections, you can use semaphores.

AmitPadhiyarOfficial, 2023-11-17:
Thank you for your reply. I tried semaphores as you mentioned, but it is not helping. The same problem occurred again. I conducted several tests, using the same code as before but removed all audio functionalities and added simple file reading functions. In short, I read two files in two different tasks without using semaphores, and it works fine. So, I decided to debug the code in your library. I removed unnecessary functions and variables from your audio.h and audio.cpp files. After removing the below-listed functions from my program, the above code works fine without adding semaphores.

My issue has been solved, but I still don't know the exact reason. I am still working on understanding what the exact reason was.

**Functions**
+ `void processWebStream();`
+ `bool readPlayListData();`
+ `const char* parsePlaylist_M3U();`
+ `const char* parsePlaylist_PLS();`
+ `const char* parsePlaylist_ASX();`
+ `const char* parsePlaylist_M3U8();`
+ `bool STfromEXTINF(char* str);`
+ `bool parseHttpResponseHeader();`
+ `inline uint32_t streamavail(){ return _client ? _client->available() : 0;}`

**Variables**
+ `WiFiClient client;`
+ `WiFiClientSecure clientsecure;`
+ `WiFiClient* _client = nullptr;`
+ `std::vector<char*> m_playlistContent;`
+ `std::vector<char*> m_playlistURL;`
+ `std::vector<uint32_t> m_hashQueue;`
+ `char* m_playlistBuff = NULL;`

AmitPadhiyarOfficial, 2023-11-24:
My program is now stable, but I don't know what the reason was. If I find the cause, I will provide an update in this issue.

closed 2023-11-24 by AmitPadhiyarOfficial