schreibfaul1/ESP32-audioI2S #611: Play audio from raw bytes

francescosalamone, 2023-11-17:
Hi, I'm working on a project that downloads an mp3 audio from web, encoded in base64, and play it.
At the moment I'm storing all data to a file, and then play it from SD Card.

Since the download take a few minutes to complete, I want to speed up the process, by playing while download, in a streaming way. 
I'm already able to read data from network by chunk, decode, and save them in a buffer, and I'm wondering if there is a way to pass the raw byte somehow to the library, or if this can be a nice feature to be implemented.

Jam-Lcz, 2024-11-18:
Have you achieved it?

KevynWithaY, 2024-11-26:
I would like this as well. Trying to figure out how to achieve that.

kilrah, 2025-01-17:
Would love to be able to play from a buffer/program memory too

tanmanh0707, 2025-02-18:
According to this, it is not possible so far
#971

closed 2025-04-04 by github-actions[bot]