schreibfaul1/ESP32-audioI2S #620: Some memory leaks in Vorbis Decoder

ma261065, 2023-12-02:
Hi,

There are a few memory leaks in the Vorbis decoder:

1) In `VORBISDecoder_FreeBuffers()`:
`vorbis_book_clear(s_codebooks);` should be `vorbis_book_clear(s_codebooks + i);`

    The way it is now the same codebook gets freed over and over again.

2) `s->q_val` is set to 0 in `vorbis_book_unpack()` which means it doesn't get freed in `vorbis_book_clear()` because it fails the `if(b->q_val)` test 

3) `s_codebooks`, `s_floor_type`, `s_residue_param`, `s_map_param` & `s_mode_param` are not freed.
They are created in `parseVorbisCodebook()`, but `s_floor_param` which is also created in `parseVorbisCodebook()` is freed in `VORBISDecoder_FreeBuffers()`

These were the ones that I found in my testing, but there may be others. Also, I guess you need to handle the case where it errors out of a function to ensure that they get freed in that case as well.

I hope this helps you to improve the decoder.

schreibfaul1, 2023-12-02:
Thanks for the error report and the support. Hope that fits so far. 
Best regards
Wolle

closed 2023-12-02 by schreibfaul1

ma261065, 2023-12-02:
Actually I just noticed that you didn't free `s_codebooks` anywhere.

ma261065, 2023-12-03:
@schreibfaul1 Thanks for the latest commit, but I think there's an error:
```
if(s_codebooks) {
        if(s_mode_param){free(s_codebooks); s_codebooks = NULL;}
}
```
Why are you checking `s_mode_param` before freeing `s_codebooks`?

I suspect it's a cut-n-paste error and you meant to write:

`if(s_codebooks){free(s_codebooks);`

Also, all of the recent free's that you added (including this one) have a double `if` e.g.

```
if(s_floor_type) {
        if(s_floor_type){free(s_floor_type); s_floor_type = NULL;}
}
```

Is that deliberate, or am I missing something?

commit 1b159418 (2023-12-02): Some memory leaks in Vorbis Decoder closed #620