schreibfaul1/ESP32-audioI2S #621: Last version error...

nodoubtman, 2023-12-10:
Hello. I'm running under Arduino IDE version 1.8.19.
Last version of ESP32-audioI2S indicate this error:

Audio audio;

'Audio' does not name a type

I'm running from the examples of the library, i didn't changed anything!

Can you help?
Thanks.
Marc.

schreibfaul1, 2023-12-10:
Hi Marc,
I think this has nothing to do with the Arduino IDE or the new master. The audio object cannot be created because the audio class is not found. It could be that the path to "audio.h" is not found or leads to a wrong destination.

nodoubtman, 2023-12-10:
> Hi Marc, I think this has nothing to do with the Arduino IDE or the new master. The audio object cannot be created because the audio class is not found. It could be that the path to "audio.h" is not found or leads to a wrong destination.

Thank you for your response.
I used Add llibrary options for adding the library.

What should i do ?

thanks.
marc.

schreibfaul1, 2023-12-10:
error: 'Audio' does not name a type' 
This happens exactly when `#include "Audio.h"` does not work. Audio.h does not exist or is somewhere else

closed 2023-12-10 by nodoubtman