schreibfaul1/ESP32-audioI2S #624: audio.connecttoFS(SD, file.name(), 100); - resuming with filePos does not work

SanZamoyski, 2023-12-12:
Hi!

I'm trying to resume flac file using audio.connecttoFS, but it starts from beggining.

also tried diffrent things, but no use:
```
//      audio.setTimeOffset(filePos);
//      audio.setAudioPlayPosition(filePos);
//      audio.setFilePos(filePos);
      audio.connecttoFS(SD, file.name(), 100);
//      delay(1000);
//      while(audio.getFilePos() == 0){
//        audio.loop();
//      }
//      audio.setAudioPlayPosition(filePos);
//      audio.setTimeOffset(filePos);
//      audio.setFilePos(filePos);
```

some file info:
```
21:32:06.230 -> Starting audio service.
21:32:06.230 -> Jumping to previous position of 146
21:32:06.230 -> info        PSRAM found, inputBufferSize: 638965 bytes
21:32:06.230 -> info        buffers freed, free Heap: 92020 bytes
21:32:06.230 -> info        Reading file: "/Eminem - The Eminem Show.flac"
21:32:06.230 -> info        FLACDecoder has been initialized, free Heap: 87644 bytes , free stack 5572 DWORDs
21:32:06.348 -> info        Content-Length: 504408175
21:32:06.489 -> info        FLAC maxBlockSize: 4096
21:32:06.489 -> info        FLAC maxFrameSize: 15809
21:32:06.489 -> info        FLAC sampleRate: 44100
21:32:06.489 -> info        FLAC numChannels: 2
21:32:06.489 -> info        FLAC bitsPerSample: 16
21:32:06.489 -> info        total samples in stream: 205087932
21:32:06.600 -> info        audio file duration: 4650 seconds
21:32:07.250 -> info        Audio-Length: 504334157
21:32:08.115 -> info        stream ready
21:32:09.644 -> info        stream ready
21:32:09.745 -> info        syncword found at pos 0
21:32:09.745 -> info        Channels: 2
21:32:09.745 -> info        SampleRate: 44100
21:32:09.745 -> info        BitsPerSample: 16
21:32:09.745 -> info        BitRate: 5397082
21:32:09.846 -> info        VBR recognized, audioFileDuration is estimated
```

Is it impossible with flac or am I doing something wrong?

Best regards!

schreibfaul1, 2023-12-13:
Hello,
FLAC is internally divided into blocks. When a jump occurs, the next block is always searched for. This is necessary so that the decoder receives the information on how to interpret the audio data of the block. 100 bytes are quite small. If, for example, your audio file is 10000000 bytes in size and you set filepos 5000000, the system jumps to the centre of the file.

SanZamoyski, 2023-12-13:
I thought it is in seconds.

Unfortunately, this:
  ```
      filePos = 10000;
      Serial.print("Jumping to previous position of ");
      Serial.println(filePos);
      audio.connecttoFS(SD, file.name(), filePos);
```

Gives the same result.
Library updated yesterday.

SanZamoyski, 2023-12-13:
Ok, 1000 was too small also!

```
filePos = 1000000;
      Serial.print("Jumping to previous position of ");
      Serial.println(filePos);
//      audio.setTimeOffset(filePos);
//      audio.setAudioPlayPosition(filePos);
//      audio.setFilePos(filePos);
      audio.connecttoFS(SD, file.name(), filePos);
```

This worked, thanks!

closed 2023-12-13 by SanZamoyski