schreibfaul1/ESP32-audioI2S #626: Clic-noise for first playback after class initialization

drtrigon, 2023-12-15:
I am working on a small project https://github.com/drtrigon/singsang. Thanks a lot for your library as it is very useful and usually working well!

As I implemented recording I need to change I2S config during runtime. ESP32-audioI2S does all the I2S config in the constructor thus I have to destroy and re-create the class.

In the beginning of the first playback I always get a quite strong clic-noise. As I figured out this is most probably related to setting the volume to some non-zero value. I found a work-a-round, as can be seen in https://github.com/drtrigon/singsang/commit/38f595246ddcc7d79ed14cdf3d3db19e1da8d177#diff-10c39dd978ada90e0d090a045c4ae5f51dad010f3b754679e15894c9b93d254bR109
basically what it does is; setting the volume to zero, play any track for some time (100 loop() iterations), stop and the finally set the desired volume:

```issue bug report:
    // prevent clic-noise when setting volume after init of audio lib (work-a-round)
    m_audio->setVolume(0);
    m_audio->connecttoFS(SPIFFS, "audio-battery.mp3");  // any file available can be used
    for (uint8_t i = 0; i < 100; ++i)  // 50 iterations should be enough, take x2
        m_audio->loop();
    m_audio->stopSong();
    m_audio->setVolume(m_currentVolume);
```

While this works as expected it feels very much like a work-a-round for some fundamental issue in the library.

I see two possible answers to this issue;

1. May be I am doing something wrong in using your libarary? In this case please tell me how to do it properly as I could not find anything related here or in the docs.
2. Please fix the bug. I can only speculate what it is, may be related to empty buffers or something else?

Thanks and regards

schreibfaul1, 2023-12-15:
It is possible that there is still (or random) data in the DMA buffer after the audio object has been created. i2s_zero_dma_buffer() should always be called. Hope this helps.

tueddy, 2023-12-15:
Hi Wolle, 
I have also heard such noises with `audio->setTimeOffset()`. Can you see if the same fix can be applied here?

Thank you very much!

tueddy, 2023-12-16:
I can confirm that the click-noise when changing the file position has been fixed with the latest commit.
As always, many thanks for the super-fast fix!

drtrigon, 2023-12-17:
I tried both, replacing my work-a-round by a `i2s_zero_dma_buffer()` as well as compiling against your commit ca85aca without success. It does not solve the issue for me.

It look like others had similar issues, confer https://github.com/earlephilhower/ESP8266Audio/issues/406

drtrigon, 2023-12-18:
update: I did a naiive try by just setting `m_i2s_config.tx_desc_auto_clear = false` but this did not help.

cziter15, 2024-02-08:
After somme digging I've found that m_filterBuff is not initialized with zeroes, which means that it will contain random content.
The filter will be populated after playing first sample.

The solution is to fill this buffer with zeroes at ctor
`memset(m_filterBuff, 0, sizeof(m_filterBuff));`

cziter15, 2024-02-08:
@drtrigon please update your library and check if the issue has been resolved

drtrigon, 2024-02-14:
> @drtrigon It looks like my PR has been merged, please update your library and check if the issue has been resolved

@cziter15: I updated to master and while I did no extensive testing it seems to work now as expected (without the work-a-round mentioned). Thanks a lot for your effort and for pinging me!

closed 2024-02-14 by drtrigon