schreibfaul1/ESP32-audioI2S #628: Uda1334 board bad audio quality

TITAS1533, 2023-12-17:
I recently started making a web radio player got esp32 and adafruit uda1334 board it works but the audio quality is awful(feels like highs non exist at all and lows kinda unnatural) compared to same speakers conected to pc and same station link playing on vlc.is that the uda1334 is garbage and need something like pcm5102 or i am doing something wrong?

code:
#include "Arduino.h"
#include "WiFi.h"
#include "Audio.h"

#define I2S_DOUT      26  // connect to DAC pin DIN
#define I2S_BCLK      27  // connect to DAC pin BCK
#define I2S_LRC       25  // connect to DAC pin LCK

Audio audio;

const char* ssid =     "SSID";
const char* password = "password";

void setup() {
    Serial.begin(115200);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) delay(1500);
    audio.setPinout(I2S_BCLK, I2S_LRC, I2S_DOUT);
    audio.connecttohost("http://s1.knixx.fm/dein_webradio_64.aac"); 
}

void loop() {
    audio.loop();
}

void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}

biologist79, 2023-12-18:
Did you connect a speaker directly to it?
If yes: please note this board does not contain an amp.

Have been working in the past with [uda1334 + TDA1308 as amp](https://forum.espuino.de/t/kopfhoererplatine-basierend-auf-uda1334-pj306b/705) and this worked fine. Meanwhile I switched to [MS6324}( https://forum.espuino.de/t/kopfhoererplatine-basierend-auf-ms6324-und-tda1308-bzw-lm4808m/1099) as it's easier to design.

For speakers I'd rather recommend MAX98357a (DAC + classD-amp).

TITAS1533, 2023-12-18:
I am conecting it to an amplifer